Hypfer/Valetudo #899: No add Button in the Timer Menu

h4ckerle, 2021-05-06:
## Describe the bug
There is no add-button shown in the Timers Menu. Works in neither Day, nor Night Theme.

## To Reproduce

1. Goto Settings
2. Click on Timers
3. See no add Button
### Screenshots

[image]
[image]

### Vacuum Model

Roborock S5
### Valetudo Version

2021.04.0/Commit 7d28ac85cea7a4081f49fe199f19adf678778f78
## Expected behavior

Add button available.
## Additional context

closed 2021-05-06 by Hypfer

h4ckerle, 2021-05-06:
HM? Why is it invalid?

Depau, 2021-05-06:
https://github.com/Hypfer/Valetudo/discussions/824

h4ckerle, 2021-05-06:
Oh. So that's intended behaviour. Thank you for sharing. Just searched the Issues but not the discussions. OK, then ill have a look at the home automation services out there.