Hypfer/Valetudo #937: Exception in attributes_message_received when handling msg

pejotigrek, 2021-05-20:
## Describe the bug
After updating to 2021.04.0 I've noticed lots of new entries in HomeAssistant's log - right now it probably didn't affected the usage of vacuum, as it is still working fine with all the automations. it's just the log growing that concerns me, and I'm not sure where to look for the solution.

As it is probably MQTT-related issue, I did tried all the steps mentioned in [Home Assistant MQTT Rewrite FAQ on the release page](https://github.com/Hypfer/Valetudo/releases/tag/2021.04.0) thinking that maybe it'll help too, but without success - entries in log persist.

Log entries pasted at the end.

## To Reproduce
1. restart HomeAssistant [software or hardware - no matter]
2. check the logs - entries show up on startup
3. wait for a random amount of time
4. check the logs - entries sometimes show up with no specific reason [meaning: HA is working as usual, vacuum is docked and charging etc.]

### Vacuum Model
Xiaomi MiJia Mi Robot Vacuum - Roborock V1

### Valetudo Version
2021.04.0

## Expected behavior
If it is Valetudo-related error - clean logs, without said entries. 

## Additional context
Log entries after HomeAssistant's restart

```
2021-05-20 16:50:31 ERROR (MainThread) [homeassistant.util.logging] Exception in attributes_message_received when handling msg on 'valetudo/1_lvn_rosita_mqtt/ZoneCleaningCapability/presets': '{"1648d7a3-9156-44b3-8a15-080f84dcc2ee":{"__class":"ValetudoZonePreset","metaData":{},"name":"Kuchnia","zones":[{"__class":"ValetudoZone","metaData":{},"points":{"pA":{"x":2955,"y":2565},"pB":{"x":3300,"y":2565},"pC":{"x":3300,"y":2675},"pD":{"x":2955,"y":2675}},"iterations":1},{"__class":"ValetudoZone","metaData":{},"points":{"pA":{"x":2985,"y":2670},"pB":{"x":3150,"y":2670},"pC":{"x":3150,"y":2740},"pD":{"x":2985,"y":2740}},"iterations":1}],"id":"1648d7a3-9156-44b3-8a15-080f84dcc2ee"},"f5e505f1-a475-4026-a53c-7fc8ea633594":{"__class":"ValetudoZonePreset","metaData":{},"name":"Korytarz","zones":[{"__class":"ValetudoZone","metaData":{},"points":{"pA":{"x":3105,"y":2740},"pB":{"x":3360,"y":2740},"pC":{"x":3360,"y":2915},"pD":{"x":3105,"y":2915}},"iterations":1}],"id":"f5e505f1-a475-4026-a53c-7fc8ea633594"},"16a36953-0b03-4d3d-b1c7-710f6825ae83":{"__class":"ValetudoZonePreset","metaData":{},"name":"Taras","zones":[{"__class":"ValetudoZone","metaData":{},"points":{"pA":{"x":2485,"y":2950},"pB":{"x":2670,"y":2950},"pC":{"x":2670,"y":3100},"pD":{"x":2485,"y":3100}},"iterations":1}],"id":"16a36953-0b03-4d3d-b1c7-710f6825ae83"}}'
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/mqtt/debug_info.py", line 32, in wrapper
    msg_callback(msg)
  File "/usr/src/homeassistant/homeassistant/components/mqtt/mixins.py", line 206, in attributes_message_received
    payload = attr_tpl.async_render_with_possible_json_value(payload)
  File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 552, in async_render_with_possible_json_value
    return _render_with_context(
  File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 1359, in _render_with_context
    return template.render(**kwargs)
  File "/usr/local/lib/python3.8/site-packages/jinja2/environment.py", line 1304, in render
    self.environment.handle_exception()
  File "/usr/local/lib/python3.8/site-packages/jinja2/environment.py", line 925, in handle_exception
    raise rewrite_traceback_stack(source=source)
  File "<template>", line 1, in top-level template code
  File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 1316, in to_json
    return json.dumps(value)
  File "/usr/local/lib/python3.8/json/__init__.py", line 231, in dumps
    return _default_encoder.encode(obj)
  File "/usr/local/lib/python3.8/json/encoder.py", line 199, in encode
    chunks = self.iterencode(o, _one_shot=True)
  File "/usr/local/lib/python3.8/json/encoder.py", line 257, in iterencode
    return _iterencode(o, 0)
  File "/usr/local/lib/python3.8/json/encoder.py", line 179, in default
    raise TypeError(f'Object of type {o.__class__.__name__} '
TypeError: Object of type LoggingUndefined is not JSON serializable
```

Log entries after random amount of time:

```
2021-05-20 16:51:53 ERROR (MainThread) [homeassistant.util.logging] Exception in attributes_message_received when handling msg on 'valetudo/1_lvn_rosita_mqtt/GoToLocationCapability/presets': '{"4b55c843-359b-4e1b-8ea7-fb231b0430cb":{"__class":"ValetudoGoToLocation","metaData":{},"name":"Kuchnia","coordinates":{"x":3198,"y":2628},"id":"4b55c843-359b-4e1b-8ea7-fb231b0430cb"}}'
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/mqtt/debug_info.py", line 32, in wrapper
    msg_callback(msg)
  File "/usr/src/homeassistant/homeassistant/components/mqtt/mixins.py", line 206, in attributes_message_received
    payload = attr_tpl.async_render_with_possible_json_value(payload)
  File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 552, in async_render_with_possible_json_value
    return _render_with_context(
  File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 1359, in _render_with_context
    return template.render(**kwargs)
  File "/usr/local/lib/python3.8/site-packages/jinja2/environment.py", line 1304, in render
    self.environment.handle_exception()
  File "/usr/local/lib/python3.8/site-packages/jinja2/environment.py", line 925, in handle_exception
    raise rewrite_traceback_stack(source=source)
  File "<template>", line 1, in top-level template code
  File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 1316, in to_json
    return json.dumps(value)
  File "/usr/local/lib/python3.8/json/__init__.py", line 231, in dumps
    return _default_encoder.encode(obj)
  File "/usr/local/lib/python3.8/json/encoder.py", line 199, in encode
    chunks = self.iterencode(o, _one_shot=True)
  File "/usr/local/lib/python3.8/json/encoder.py", line 257, in iterencode
    return _iterencode(o, 0)
  File "/usr/local/lib/python3.8/json/encoder.py", line 179, in default
    raise TypeError(f'Object of type {o.__class__.__name__} '
TypeError: Object of type LoggingUndefined is not JSON serializable
```

Hypfer, 2021-05-20:
This is not a bug in Valetudo but rather some oddity in the Logger of Home Assistant.

Please file an issue in home-assistant/core as I can do absolutely nothing about this

closed 2021-05-20 by Hypfer

pejotigrek, 2021-05-20:
ok, thanks for clarifying that.

JanickGers85, 2021-08-05:
@pejotigrek @Hypfer 
Hi everyone! I got the same problem and found a solution, I followed these steps:
- Put a tick on _Delete autodiscovery on shutdown_ (home assistant autodiscovery)
  
[image]

- Shutdown vacuum with physical button (I own a Xiaomi first generation)
- Delete _valeduto_ and valetudo's _homeassistant_* related topic from the broker if any remains (latest one are for autodiscovery)
[image]
 *actually I've removed the whole homeassistant topic because valetudo was the only subtopic there
- power on your vacuum, autodiscovery should kick in and all log errors should be gone! (at least it worked for me)

I've noticed that i have all sensors and camera duplicated and one of each pair was unavailable, doing so cleared all up and no more errors on log.
Hope it helps! Valetudo rocks :)

Depau, 2021-08-05:
This should probably be in the Troubleshooting/Home assistant page since most people aren't going to read the release notes of 4 releases ago.

JanickGers85, 2021-08-05:
Since I can't reply [here](https://github.com/home-assistant/core/issues/50905) (conversation has been locked) I thought it would be a nice thing to share it on this issue so if someone search for it or is following this thread could found it useful.
I don't know where is the troubleshooting/HA page is, if you know can you please tell me so I'll share this procedure there? TY!

Depau, 2021-08-05:
https://github.com/Hypfer/Valetudo/tree/master/docs/_pages

If you go to the releases page and find the release in which the new MQTT functionality was added there's a home assistant troubleshooting section that you can copy paste into a pull request.