Hypfer/Valetudo #940: Valetudo do not updated vacuum entity and state on HA

JanickGers85, 2021-05-22:
## Describe the bug
Sometime the robot stop sending update to HA: it just remain in the current state (docked, cleaning, whatever) until I reboot it or goes to UI and reconnect to MQTT broker without changing any settings.
If I send a MQTT to the topic to start cleaning, or locate the robot, it respond to command.

## To Reproduce

It happens quiet randomly, can't say how to reproduce.
I tried to force a MQTT reconnect via a HTTP PUT at 5:00AM to have the robot up and running for the morning cleaning but this didn't workaround'ed the problem

### Screenshots

It ended cleaning and it's docked, but HA still report the cleaning state (no text shown beside "Stato", BTW)
[image]

### Vacuum Model

Xiaomi Mi Robot Vacuum (1st gen)
### Valetudo Version

2021.04.0
## Expected behavior

Report correct info to HA (via MQTT).

## Additional context

Here the logs: I've added a few more info between <<<>>>

```
[2021-05-22T01:07:25.740Z] [INFO] Set Logfile to /tmp/valetudo.log   <<<Built-in daily reboot>>>
[2021-05-22T01:07:25.751Z] [INFO] Autodetected RoborockV1ValetudoRobot
[2021-05-22T01:07:25.976Z] [INFO] Starting Valetudo 2021.04.0
[2021-05-22T01:07:25.978Z] [INFO] Commit ID: 7d28ac85cea7a4081f49fe199f19adf678778f78
[2021-05-22T01:07:25.979Z] [INFO] Configuration file: /mnt/data/valetudo/valetudo_config.json
[2021-05-22T01:07:25.980Z] [INFO] Logfile: /tmp/valetudo.log
[2021-05-22T01:07:25.982Z] [INFO] Robot: Beijing Roborock Technology Co., Ltd. Xiaomi Mi Robot Vacuum (RoborockV1ValetudoRobot)
[2021-05-22T01:07:25.983Z] [INFO] JS Runtime Version: v14.16.1
[2021-05-22T01:07:25.984Z] [INFO] Arch: arm
[2021-05-22T01:07:25.986Z] [INFO] Max Heap Size: 34 MiB
[2021-05-22T01:07:25.987Z] [INFO] Node Flags: --expose-gc --max-heap-size=34
[2021-05-22T01:07:26.016Z] [INFO] DeviceId 260856367
[2021-05-22T01:07:26.017Z] [INFO] IP 127.0.0.1
[2021-05-22T01:07:26.019Z] [INFO] CloudSecret ***
[2021-05-22T01:07:26.021Z] [INFO] LocalSecret ***
[2021-05-22T01:07:26.540Z] [INFO] Dummycloud is spoofing 127.0.0.1:8053 on 127.0.0.1:8053
[2021-05-22T01:07:26.550Z] [INFO] Webserver running on port 80
[2021-05-22T01:07:26.570Z] [INFO] Map Upload Server running on port 8079
[2021-05-22T01:07:30.704Z] [INFO] Connected successfully to MQTT broker
[2021-05-22T01:07:31.561Z] [WARN] Error during time sync: Error: getaddrinfo EAI_AGAIN pool.ntp.org
    at GetAddrInfoReqWrap.onlookup [as oncomplete] (dns.js:67:26) {
  errno: -3001,
  code: 'EAI_AGAIN',
  syscall: 'getaddrinfo',
  hostname: 'pool.ntp.org'
}
[2021-05-22T01:07:32.217Z] [INFO] MQTT configured
[2021-05-22T01:07:33.581Z] [INFO] Cloud connected
[2021-05-22T01:08:33.004Z] [INFO] Successfully set the robot time via NTP to 2021-05-22T01:08:33.891Z
[2021-05-22T03:00:01.742Z] [INFO] Connected successfully to MQTT broker <<<My MQTT forced reconnect as a workaround>>>
[2021-05-22T03:00:02.802Z] [INFO] MQTT configured
[2021-05-22T06:39:16.795Z] [INFO] Cloud message timed out. Assuming that we're not connected anymore
[2021-05-22T06:39:19.065Z] [INFO] Cloud connected
[2021-05-22T06:41:18.220Z] [INFO] Cloud message timed out. Assuming that we're not connected anymore
[2021-05-22T06:41:18.233Z] [WARN] Failed to get consumables: MiioTimeoutError: request timed out:{"method":"get_consumable","params":[],"id":971}
    at Timeout.timeout [as _onTimeout] (/snapshot/Valetudo/lib/miio/MiioSocket.js:197:32)
    at listOnTimeout (internal/timers.js:554:17)
    at processTimers (internal/timers.js:497:7)
[2021-05-22T06:41:35.772Z] [INFO] Cloud connected
[2021-05-22T06:46:50.046Z] [INFO] Set Logfile to /tmp/valetudo.log
[2021-05-22T06:46:50.058Z] [INFO] Autodetected RoborockV1ValetudoRobot
[... 277 lines cut ...]
    at RobotMqttHandle.configure (/snapshot/Valetudo/lib/mqtt/handles/MqttHandle.js:167:29)
    at processTicksAndRejections (internal/process/task_queues.js:93:5)
    at async /snapshot/Valetudo/lib/mqtt/handles/RobotMqttHandle.js:110:13
    at async MqttController.reconfigure (/snapshot/Valetudo/lib/mqtt/MqttController.js:497:13)
    at async RobotMqttHandle.onStatusAttributeEvent (/snapshot/Valetudo/lib/mqtt/handles/RobotMqttHandle.js:97:9)
[2021-05-22T08:03:32.231Z] [WARN] MQTT handle valetudo/pippo/BatteryStateAttribute failed to configure Error: Handle may only be configured while the MQTT controller is not initialized
    at BatteryStateMqttHandle.configure (/snapshot/Valetudo/lib/mqtt/handles/MqttHandle.js:160:19)
    at BatteryStateMqttHandle.configure (/snapshot/Valetudo/lib/mqtt/handles/RobotStateNodeMqttHandle.js:30:21)
    at RobotMqttHandle.configure (/snapshot/Valetudo/lib/mqtt/handles/MqttHandle.js:167:29)
    at processTicksAndRejections (internal/process/task_queues.js:93:5)
    at async /snapshot/Valetudo/lib/mqtt/handles/RobotMqttHandle.js:110:13
    at async MqttController.reconfigure (/snapshot/Valetudo/lib/mqtt/MqttController.js:497:13)
    at async RobotMqttHandle.onStatusAttributeEvent (/snapshot/Valetudo/lib/mqtt/handles/RobotMqttHandle.js:97:9)
[2021-05-22T08:04:07.006Z] [INFO] Successfully set the robot time via NTP to 2021-05-22T08:04:07.891Z
[2021-05-22T08:11:12.978Z] [INFO] Cloud message timed out. Assuming that we're not connected anymore
[2021-05-22T08:11:14.260Z] [WARN] Failed to get consumables: MiioTimeoutError: request timed out:{"method":"get_consumable","params":[],"id":365}
    at Timeout.timeout [as _onTimeout] (/snapshot/Valetudo/lib/miio/MiioSocket.js:197:32)
    at listOnTimeout (internal/timers.js:554:17)
    at processTimers (internal/timers.js:497:7)
[2021-05-22T08:11:31.421Z] [INFO] Cloud connected
[2021-05-22T08:15:17.428Z] [INFO] Unknown cloud message received: {"method":"event.back_to_origin_fail","params":[0],"id":742344196}
[2021-05-22T10:31:26.220Z] [INFO] Connected successfully to MQTT broker <<<My manual MQTT reconnect to bring robot to life again>>>
[2021-05-22T10:31:27.120Z] [INFO] MQTT configured

```

Depau, 2021-05-23:
Fixed in next release, and this is the 4th time i close reports of this exact same issue.

There's a search bar, please use it.

closed 2021-05-23 by Depau

JanickGers85, 2021-05-24:
Sorry, actually I searched on issues and I didn't find any other one similar to mine.

Great to know that it will be fixed in next release and thanks for the great work.