Hypfer/Valetudo #957: Xiaomi Mi Robot Vacuum reseting/crashing during cleaning

jkalousek, 2021-05-30:
**Describe the bug**
I originally installed Valetudo v 2021.03.0 and recently upgraded to 2021.04.0. I do run MQTT with Home Assistant. After installation I noticed that sometimes after we got home, robot was just sitting with low battery in middle of the room. It didn't happened every time just sometimes and usually after cleanup was resumed (after charge from low battery state). This was happening for about a month (not always). After that I decided to try to disable MQTT and for about a week the problem didn't occur again so I re enabled MQTT again and installed new Valetudo version (2021.04.0) and we were fine for about next 14 days and today it happened again except this time I had a chance to watch the behavior on my own eyes + on the map.

This is what is happening based on my observation:
Vacuum is cleaning (at this point for about an hour+ it already did recharge once.)
Suddenly it will lost connection in Web GUI and Vacuum itself goes quiet.
Few seconds after that it suddenly starts vacuuming again and Web GUI comes back with following:
It knows original map so I assume that it knows that it should resume where it left of but timer starts over again and most importantly it thinks that it is now in the dock, not where it "died", which of course thrown everything off as it is not able to get back on track.

From now on this will start happening every 50-120 seconds over and over again until it dies because of low battery.

Based on logs It seams that it just either crashed or just resets:
```
[2021-05-30T11:25:41.347Z] [INFO] Set Logfile to /tmp/valetudo.log
[2021-05-30T11:25:41.369Z] [INFO] Autodetected RoborockV1ValetudoRobot
[2021-05-30T11:25:41.697Z] [INFO] Starting Valetudo 2021.04.0
[2021-05-30T11:25:41.700Z] [INFO] Commit ID: 7d28ac85cea7a4081f49fe199f19adf678778f78
[2021-05-30T11:25:41.701Z] [INFO] Configuration file: /mnt/data/valetudo_config.json
[2021-05-30T11:25:41.703Z] [INFO] Logfile: /tmp/valetudo.log
[2021-05-30T11:25:41.705Z] [INFO] Robot: Beijing Roborock Technology Co., Ltd. Xiaomi Mi Robot Vacuum (RoborockV1ValetudoRobot)
[2021-05-30T11:25:41.706Z] [INFO] JS Runtime Version: v14.16.1
[2021-05-30T11:25:41.707Z] [INFO] Arch: arm
[2021-05-30T11:25:41.711Z] [INFO] Max Heap Size: 34 MiB
[2021-05-30T11:25:41.712Z] [INFO] Node Flags: --expose-gc --max-heap-size=34
[2021-05-30T11:25:41.726Z] [INFO] DeviceId 74206929
[2021-05-30T11:25:41.728Z] [INFO] IP 127.0.0.1
[2021-05-30T11:25:41.758Z] [INFO] CloudSecret REDACTED
[2021-05-30T11:25:41.760Z] [INFO] LocalSecret REDACTED
[2021-05-30T11:25:42.417Z] [INFO] Dummycloud is spoofing 127.0.0.1:8053 on 127.0.0.1:8053
[2021-05-30T11:25:42.427Z] [INFO] Webserver running on port 80
[2021-05-30T11:25:42.450Z] [INFO] Map Upload Server running on port 8079
[2021-05-30T11:25:47.436Z] [WARN] Error during time sync: Error: getaddrinfo EAI_AGAIN pool.ntp.org
    at GetAddrInfoReqWrap.onlookup [as oncomplete] (dns.js:67:26) {
  errno: -3001,
  code: 'EAI_AGAIN',
  syscall: 'getaddrinfo',
  hostname: 'pool.ntp.org'
}
[2021-05-30T11:25:53.901Z] [INFO] Cloud connected
[2021-05-30T11:25:55.488Z] [INFO] Connected successfully to MQTT broker
[2021-05-30T11:25:59.570Z] [INFO] MQTT configured
[2021-05-30T11:25:59.659Z] [INFO] Cloud message timed out. Assuming that we're not connected anymore
[2021-05-30T11:26:14.135Z] [INFO] Cloud connected
[2021-05-30T11:26:15.678Z] [WARN] Token is okay, however we're unable to reach the vacuum { retries: 10, method: 'get_consumable', args: [] }
[2021-05-30T11:26:50.006Z] [INFO] Successfully set the robot time via NTP to 2021-05-30T11:26:50.892Z
[2021-05-30T11:27:23.106Z] [WARN] Failed to parse uploaded map
[2021-05-30T11:27:25.109Z] [WARN] Failed to parse uploaded map
[2021-05-30T11:27:27.115Z] [WARN] Failed to parse uploaded map
[2021-05-30T11:27:34.283Z] [WARN] Failed to parse uploaded map
[2021-05-30T11:30:19.087Z] [INFO] Cloud message timed out. Assuming that we're not connected anymore
[2021-05-30T11:30:37.072Z] [INFO] Cloud connected
```

It might be just coincidence that this started happening after I flashed with Valetudo but I'm not aware that I would see same error in past 3 years that we are using this Vacuum.

EDIT:
If I stop the cleaning and start it again from scratch it starts vacuuming normally.

**Vacuum Model:** Xiaomi Mi Robot Vacuum (Gen1)

**Valetudo Version:** 2021.04.0

Hypfer, 2021-05-30:
> Vacuum is cleaning (at this point for about an hour+ it already did recharge once.)

How many m² are we talking about here?

jkalousek, 2021-05-30:
> 
> 
> > Vacuum is cleaning (at this point for about an hour+ it already did recharge once.)
> 
> How many m² are we talking about here?

About +/- 130 m² the battery is as I stated about 3y old so it definitely lost capacity and we are using it on High power preset.

Hypfer, 2021-05-30:
Usually, these symptoms point to memory (as in RAM) issues, which is likely also the issue here.

It is unclear if it's
- a bug in Valetudo
- something that could be mitigated using different approaches somewhere

or
- just related to the map size in general with no way around it

You're seeing MQTT having some influence on it, because to publish the mqtt message, the map data has to be serialized to a string which at 130m² most likely takes up multiple megabytes of memory.

Unfortunately, my floorplan is less then half the size of yours so I can merely guess what is going on since it's not happening here

jkalousek, 2021-05-30:
Is there any way we can debug this? I assume that os that is running on the vacuum is read only so during reset it will lost all logs?

Hypfer, 2021-05-30:
Check out https://github.com/Hypfer/Valetudo/issues/712 which is about the S5 Max but also the same issue. It contains debugging instructions

Context:
The S5 Max features only 256mb of ram as opposed to the 512mb of the V1, however roborock optimized its software to deal with that just fine. Unfortunately, they've never ported that back to the V1.
The did port it to the S5 (non-max) though so thats why were not seeing memory issues there anymore even with huge maps

I think I'll try streaming the map data serialization. That could improve things. We'll see

jkalousek, 2021-05-30:
Could there maybe also be an option in settings that would disable map related operations all together? I actually do not need the map at all I just want to see vacuum status and have a option to Start/Stop/RTH from my HA.

Hypfer, 2021-05-30:
You could disable "provide map data" in the mqtt options

aHVzY2g, 2021-05-30:
I had the same struggle(I think) with 2021.04.0 I updated now to 2021.05.0 and will see if that fixes the problem.

Hypfer, 2021-05-30:
> I think I'll try streaming the map data serialization. That could improve things. We'll see

https://github.com/Hypfer/Valetudo/commit/ca95ac94b5ab47625d8ef16f156118f8a84ff00c

Built binaries can be found here:

https://github.com/Hypfer/Valetudo/runs/2704846191

Please try and see if it makes any difference. It only changes the mqtt map publishing so if you've turned that off, there should be no difference at all

jkalousek, 2021-05-30:
I will try for a start to disable the map data in MQTT as I didn't notice this option before and also watch the memory during some of future vacuuming. I will post feedback as soon as I get to it. 

Just an Idea... as we know this is a problem which will probably not be fixed in core and we can check for memory stats wouldn't it be best to actually check for memory status and in case that we will get under some % of free ram just skip the map serialization and just send basic mqtt data + report this fact in log? I see map only as additional benefit but the main function of machine should take priority before "comfort" functions in my opinion.

ManuelW77, 2021-06-05:
The same here. No Map is shown and the robot is rebooting.
I have mqtt completely disabled.

Hypfer, 2021-07-03:
Closing due to inactivity. Please open a new one if this is still an issue in the newest version

closed 2021-07-03 by Hypfer

mhmilo24, 2022-01-18:
For future purposes:
This seems to be a problem caused by a faulty lidar unit. It's either a dying motor or ageing rubber band within the unit. No need to fix the software, except if you could measure the lidar spin and warn users preemptively.