Hypfer/Valetudo #958: Zone cleanup not working after upgrading to 2021.05.0

luke-barnett, 2021-05-31:
## Describe the bug
Unable to send zone commands after upgrading to 2021.05.01

## To Reproduce

1. Open web UI
2. Click on zone
3. Select zone
4. See error

### Screenshots

[image]

[image]

### Vacuum Model
Roborock S5

### Valetudo Version
2021.05.0

## Expected behavior

The vacuum starts a zone clean

## Additional context

Also doesn't work via MQTT API.

Restarted the vacuum with no sucess.

Hypfer, 2021-05-31:
#952

> Also doesn't work via MQTT API.

Wrong.

closed 2021-05-31 by Hypfer

luke-barnett, 2021-05-31:
@Hypfer apologies, I thought searching based on the API endpoint giving a 404 would tell me if it was logged already.

I've got it all set up again by directly calling the MQTT publish commands needed. I initially assumed the MQTT API was broken as the home assistant vacuum.send_command service wasn't working and that is all auto-discovered via MQTT.

Is there an updated format to pass through the vacuum.send_command service or should I just leave it with an MQTT publish?

Thanks for all the hard work. Appreciate it.

Hypfer, 2021-05-31:
>Is there an updated format to pass through the vacuum.send_command service

Nope. That was not reimplemented because it interfered with the vendor-agnostic mqtt rewrite. I'm sure it would've been possible somehow to accomodate for that but given that using `mqtt.publish` achives the same thing with very little changes required, the decision against it was made.

Considering the state `vacuum.mqtt` is in and how many things such as e.g. zone presets, water grade control etc don't fit in there very well or even at all, it makes more sense to ditch that and write a valetudo `custom_component` which could then provide nice `valetudo.something` services

luke-barnett, 2021-05-31:
Interesting. Thanks for the insight.