Hypfer/Valetudo #977: Manuel Control is gone

valknut89, 2021-07-04:
Since the "new" release names Manuals control is gone. I use the actual prebuild firmware, which is linked in instructions

closed 2021-07-04 by valknut89

rmickael62, 2021-07-10:
Did you find the manual control of valetudo? because I can't find it either

valknut89, 2021-07-12:
>  [data:image/png;base64,iVBORw0KGgoAAAANSUhEUgAAAD8AAAA/CAYAAABXXxDfAAAAAXNSR0IArs4c6QAAAARzQklUCAgICHwIZIgAAABcSURBVGgF7dABDQAAAMKg909tDwcRKAwYMGDAgAEDBgwYMGDAgAEDBgwYMGDAgAEDBgwYMGDAgAEDBgwYMGDAgAEDBgwYMGDAgAEDBgwYMGDAgAEDBgwYMGDgZGA+QwABeHisFQAAAABJRU5ErkJggg==###24x24:true###][Verfolgungsbild][https://github.com/notifications/beacon/APQYSTDWNBVK2MH2DIP3ZE3TXAKI3A5CNFSM47Y7UJJKYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOGRHUCAY.gif]

No, but I read in FAQ that the function maybe will come back in one of the next releases.
Didn't Check the july release yet.