Hypfer/Valetudo #986: MQTT Monitoring

Hypfer, 2021-07-13:
Similar to how it is done in the NTPClient, the MQTTController should provide means of fetching its state + other health-related information.
After the state-machine rework is done, this should be pretty much a no-brainer.

Also, it would be neat to count received messages by topic for debug purposes etc.
See also: https://stackoverflow.com/a/13298258/10951033

@Depau any thoughts?

Depau, 2021-07-13:
I barely followed the recent developments from the release notes, and my vacuum is still running whatever last build I built before I basically disappeared.

With that said I did read something about the new health check endpoints (which sounds good and very useful for viomi since it likes to die occasionally).

I'm not entirely sure, though, why you'd want to monitor message counts (doesn't mosquitto do that on its own?). IDK what else you could otherwise monitor other than the state machine state since if MQTT is broken and MQTT.js didn't tell us there's not a lot to do other than open up the code and implement a workaround. It could make sense to monitor the last failure reason though I don't recall MQTT.js's error messages to be particularly useful.

With state-machine rework I think you mean something like what I had proposed in the group, right?

Depau, 2021-07-13:
By the way, lately I've been pretty much ignoring all GitHub notification and I was about to dismiss it, I ended up here only because I was curious on what was going on on the MQTT side.

If you need my feedback it's probably more effective to hard-ping me on telegram ;) sorry if I missed any other tags here, feel free to send them back.

closed 2022-02-12 by Hypfer