Hypfer/Valetudo #987: More/Better Swagger Examples

Hypfer, 2021-07-13:
The swagger docs could use some more examples for common things to do such as ordered Segment Cleaning as well as additional proof-reading.

The actual spec is built from many small json files spread all around the project in `doc` folders via the `build_openapi_schema` npm script to keep them readable.
e.g.
https://github.com/Hypfer/Valetudo/blob/master/backend/lib/webserver/capabilityRouters/doc/MapSegmentationCapabilityRouter.openapi.json

kcindric, 2021-07-30:
Could you write a list of endpoints which could use additional examples?

Hypfer, 2021-07-30:
Actually basically all of them. Only the TimerRouter has some examples

The rest is just infered from the schema by the swagger ui

kcindric, 2021-07-30:
Ok, I'll try to commit to this. What a better way to learn a project than go through each and every endpoint :)

PR #1068 (2021-09-05): Swagger examples for ManualControlCapability
## Type of change

Type A:
- [ ] Bug fix (non-breaking change which fixes an issue)
- [ ] Refactor/Code Cleanup
- [ x] Docs
- [ ] Capability implementation for existing core capability
- [ ] New robot implementation
  
Type B:
- [ ] New capability
- [ ] New core feature

# Description (Type A)
ManualControlCapability had now swagger examples, the generated example value was wrong
 
Fixes #987
Fixes #1065
closed 2022-11-29 by Hypfer