Hypfer/Valetudo #988: "Zones" not shown anymore

tetzlav, 2021-07-14:
## Describe the bug
After update from 2021.04.0 to 2021.06.0 "Zones" not shown in "Home" page anymore.

## To Reproduce
If i copy back the 2021.04.0 binary and restart the "Zones" are shown again.

### Screenshots
2021.04.0
[image]
[image]
2021.06.0
[image]

### Vacuum [Model](url)
Roborock S5 Max### Valetudo Version

### Config files
[valetudo_config.json.txt](https://github.com/Hypfer/Valetudo/files/6814490/valetudo_config.json.txt)

closed 2021-07-14 by Hypfer

tetzlav, 2021-07-14:
any more info why closed?

Hypfer, 2021-07-14:
https://github.com/Hypfer/Valetudo/discussions/968#discussioncomment-919167

https://github.com/Hypfer/Valetudo/issues/952