Hypfer/Valetudo #990: `api/v2/robot/capabilities/PersistentMapControlCapability` server error

thetic, 2021-07-18:
## Describe the bug

`api/v2/robot/capabilities/PersistentMapControlCapability` REST endpoint returns 500.

## To Reproduce

```sh
$ curl -X 'PUT' "$ROBOT_IP/api/v2/robot/capabilities/PersistentMapControlCapability'" -H 'accept: */*' -H 'Content-Type: application/json'  -d '{"action": "enable"}'  --fail-with-body
curl: (22) The requested URL returned error: 500
"The Lab Status is same"%
```

### Screenshots

N/A

### Vacuum Model

Roborock S5 Max

### Valetudo Version

2021.06 (8eae1786c9d85a10c3222144e1a9561141c10746)

## Expected behavior

200 code (as documented in swagger)

## Additional context

Hypfer, 2021-07-18:
Basically all capability routers might return 500 if the execution of whatever they're doing fails.
The reason for why this isn't documented is time. It's simply missing in the swagger docs (see also #987)

In this case, `The Lab Status is same` is the response error message by the robot's firmware which could use a "Error from Robot:" prefix or something like that. The S5 Max is _weird_ (as in: not reverse-engineered properly) in that regard and therefore requires special treatment:

```
Sören, [01.06.21 13:01]
Persistent maps need to be enabled and also exist. For some reason, that seems to break for people.

To fix this, please try the following

- ssh onto the robot and change /mnt/data/rockrobo/lab.cfg to 0
- reboot
- re-enable persistent maps in the valetudo ui
- do a full cleanup
```

closed 2021-07-18 by Hypfer

thetic, 2021-07-18:
OK, to make sure I understand:
- the 500 error is expected, but undocumented (captured in #987)
- the s5max capability routers are still incomplete (i can't find an existing ticket)

Hypfer, 2021-07-18:
Yes and no

Yes, the error is expected but undocumented
No they are complete but the whole persistent map thing works a bit differently on the s5 max and newer and therefore requires some extra work by the user which maybe could be solved by valetudo but idk

thetic, 2021-07-18:
Got it. this endpoint is simply not expected to work on the s5max. thanks for your attention.