Hypfer/Valetudo #991: iobroker.mqtt support?

tetzlav, 2021-07-18:
## Describe the bug
Maybe its a problem of iobroker.mqtt, but i'am not able to send commands from iobroker.mqtt.

## To Reproduce
Try to set `mqtt.0.valetudo.robot.BasicControlCapability.operation.set` with payload `START` 
or start zone cleaning via `mqtt.0.valetudo.robot.ZoneCleaningCapability.start.set` with payload `["26b79408-c938-46bc-9b4a-db400ed82c77"]`.

If i use mosquitto like this
```
# mosquitto_pub -h 192.168.XXX.YYY -t 'valetudo/robot/ZoneCleaningCapability/start/set' -m '["26b79408-c938-46bc-9b4a-db400ed82c77"]'
# mosquitto_pub -h 192.168.XXX.YYY -t 'valetudo/robot/BasicControlCapability/operation/set' -m 'STOP'`
```
everthing works like expected.

### Vacuum Model
Roborock S5 Max

### Valetudo Version
2021.6.0

## Expected behavior
Start cleaning

## Valetdo log
```
[2021-07-18T20:31:14.360Z] [WARN] Received a retained MQTT message. This is almost certainly an error. Discarding. {
  topic: 'valetudo/robot/BasicControlCapability/operation/set',
  message: 'START'
}
[2021-07-18T20:31:41.907Z] [WARN] Received a retained MQTT message. This is almost certainly an error. Discarding. {
  topic: 'valetudo/robot/ZoneCleaningCapability/start/set',
  message: '["26b79408-c938-46bc-9b4a-db400ed82c77"]'
}
```

Hypfer, 2021-07-18:
> If i use mosquitto like this everthing works like expected.

This is not the iobroker issue tracker.

closed 2021-07-18 by Hypfer

tetzlav, 2021-07-18:
I understand your point. Because of this i wrote, MAYBE thats a problem of iobroker.mqtt.
But the valetudo log says that topic and message are correct and conform with the docs...

And if i open an issue in iobroker.mqtt repo, what should i write whats going wrong? The message is correct transfered but not understood!? Where i should start to debug? Im no software developer.

Best regards

Hypfer, 2021-07-18:
> Where i should start to debug? 

You could start by actually reading the error message.

`Received a retained MQTT message. This is almost certainly an error. Discarding.`

If you don't know what a retained MQTT message is, a quick google search for "mqtt retained" returns 71.000 results with very good explainations found directly on page one.

But honestly you don't even need to know what retained means to deduce that the message is retained but should not be retained as Valetudo discarded it for this reason.
The real question you're trying to answer is "How do I send MQTT messages via iobroker without them being considered retained?".

Zero developer knowledge required.
This is just simple problem analysis where the actual subject could be anything.
This could've been about horses.

Hypfer, 2021-07-19:
I'd like to further point out that what you've been experiencing is a failsafe designed to work around common user errors.

With retained messages for command topics, the robot will start cleaning every time it reboots which is every day at 3am as it will receive the command every time it connects to the mqtt broker.

There might be some "always retain" flag or something in the iobroker. I have no idea.
I only know that their integrated mqtt broker is not compliant to spec and therefore did already cause issues before

Depau, 2021-07-19:
It might be worth specifying better in the error message that the problem is on the chair, not in the computer.

I.e. "Most certainly you or the home automation software integration you are using are sending the MQTT command incorrectly. Please remove the "retained" flag to fix this issue."

Hypfer, 2021-07-19:
@Depau good idea!
7e28c663bf4e321c80350df932e7914c4901ca29

Depau, 2021-07-19:
(you also copy pasted my typos, lol)