Hypfer/Valetudo #1006: S5 -- 2021.07.01 - Periodic reboots, lost connection, can no longer ssh/no wifi

yieldhog, 2021-08-06:
## Describe the bug

I've been using Valetudo 2 years, Xiaomi S5. Upgraded with full flash of latest version 2021.07.01. Since 2021.06 I have found the vacuum periodically will lose connection to MQTT and wifi, when it connects, I cannot login to the web interface, but can SSH in. It has been pretty unstable over the past few months whereas before the big change release I never had an issue.

However, as of today, the vacuum no longer has any connection to wifi, I cannot SSH in at all. There has been no changes to my network.

Is there a way to "reboot" the vacuum manually without resetting it to factory and refreshing if it is unable to connect to wifi? I am at a loss here on what to do.

I upgraded using step by step dustbuilder tutorial successfully. Thank you

## To Reproduce
N/A

### Vacuum Model

Roborock S5

### Valetudo Version

2021.07.01

closed 2021-08-06 by Hypfer

yieldhog, 2021-08-06:
@Hypfer I’d love to know why this is invalid. I found a bug a year ago and I’ve seen you rip people to shreds who found valid issues in the past. Nice.

Hypfer, 2021-08-06:
[image]

Hmm why could this be invalid. Surely because I like to roast people with valid issues because I'm a terrible person.

Also, lets ignore the 95% garbage bug reports caused by user error and just look at the one time where that wasn't the case.
I'm sure that that's a valid representation of what I'm doing.