Hypfer/Valetudo #1012: Returning home during cleaning stage requires pausing first

tlpbu, 2021-08-07:
## Describe the bug
If any vacuum is perfoming cleaning, return home mqtt command is ignored unless paused inbetween.

## To Reproduce
Home Assistant automation that has worked for years but now is just not doing anything without any errors logged
```
- alias: Send Servantess to dock when someone arrives home during repeated cleanup
  trigger:
    - platform: state
      entity_id: group.family
      to: 'home'
  condition:
...
  action:
    - service: vacuum.return_to_base
      entity_id: vacuum.servantess
```

### Vacuum Model
RoborockV1ValetudoRobot

### Valetudo Version
2021.07.1

## Expected behavior
React on BasicControlCapability mqtt command HOME without pausing vacuum.

## Additional context
This was working for years until I upgraded Valetudo from 2021.06.2 to 2021.07.1. So behaviour has changed in later relases.

Hypfer, 2021-08-07:
No.

There were absolutely no changes to that

closed 2021-08-07 by Hypfer

tlpbu, 2021-08-07:
There were actually huge changes related to MQTT implementation according to release notes.
Just noticed the same behavior is is in web UI as well - Home button is disable during cleaning and first requires pausing.
Similar weirdness is that Stop button presses in vacuum web UI are ignored during Returning state.

Hypfer, 2021-08-07:
We still send exactly the same commands to the robot.
What its firmware does with them is out of our control.