Hypfer/Valetudo #1019: Improve Dreame Status Code Handling

Hypfer, 2021-08-14:
Currently, the Dreame implementation only uses the WorkMode + Error code to determine the status.

This however is wrong, as it's also important to check, whether or not the robot is docked, which is something only available in the map data.
A quite weird implementation tbh.

When left alone, the WorkMode may change to some Powersave state which is currently mapped to docked, even though it's in fact not docked. To fix that, we need to take e.g. the Map Data into account.

rimrul, 2021-08-14:
If i understand the issue correctly, the robot enters some powersaving state after a while (probably only when not cleaning) and that state needs to be mapped to multiple valetudo states depending on context. Do we need to handle the mapdata for this or could we just cache the last known state that cleanly maps to one valetudo state without context 
clues and derive our current state from that?

I.e. if we know the robot wasn't docked before entering powersaving, can we assume it doesn't become docked without leaving powersaving?

Hypfer, 2021-08-15:
Good idea, however I noticed that I'm just stupid.

The Battery service has another property beside the charge %: The on-charger state.

Its value being one of
1 = On Charger
2 = Not on Charger
5 = Returning to charger

and maybe more?
Anyways. Much easier to implement

Hypfer, 2021-08-15:
Fixed in 343c749a

closed 2021-08-15 by Hypfer