Hypfer/Valetudo #1050: Current map is null

AdriaanDeVos, 2021-08-29:
## Describe the bug

Forbidden Markers functionality does not work anymore.

## To Reproduce

1. Go to `Zones`
2. Go To `Configure Forbidden Zones`
3. Press the `save` button
4. Error `current map is null` occurs

### Screenshots

[image]

### Vacuum Model

Roborock S5 Max

### Valetudo Version

2021.08.0

## Expected behavior

I expect to be able to save the changes I have made in the `Forbidden Markers` screen.

## Additional context

In addition to this error I am also unable to select any segments in the `Edit Segments` menu. Therefore, I can't use this functionality too. I think these problems might be related. These problems started to occur after I reset my map with the following steps:
1. Go to `Settings`
2. Go to `Persistent Data`
3. Enable the "Enabled" checkbox
4. Press `Reset Map`

Before resetting my map I had no issues whatsover with the functionality so I think it might be related.  
Feel free to let me know if you need any logs or I can help testing out anything.

Hypfer, 2021-08-29:
[image]

closed 2021-08-29 by Hypfer

AdriaanDeVos, 2021-08-29:
Thanks for your quick response Hypfer. I am unable to find any solution for this on the valetudo.cloud website or on the discussions on GitHub. If you know that this is not a bug, could you link me to a possible solution?

Hypfer, 2021-08-29:
I think you're misjudging this situation.

If you're opening a bug report, you must provide strong evidence that something is a Bug in Valetudo.
It is not _my_ job to provide proof that something isn't.

AdriaanDeVos, 2021-08-29:
I understand that it is not your job to provide proof that it is not a bug.  

I just thought this was a bug because it is unexpected behaviour within Valetudo that is not yet documented somewhere resulting in a major part of the functionality not working because the map is not correctly found/registered...

Hypfer, 2021-08-29:
The message you're seeing is provided by the robots firmware and tells you that it has no map.
If you give it a map it will start working again.

This should've been a support thread

AdriaanDeVos, 2021-08-29:
I understand, sorry for my mistake. I have created a support thread in the hope that this can be resolved. :)