Hypfer/Valetudo #1063: MQTT with SSL not working anymore

darootler, 2021-08-31:
## Describe the bug
I just upgraded valetudo from 2021.08.0 to 2021.08.1 and MQTT with SSL is not working anymore. I did no config changes after the upgrade.

## To Reproduce
Configure MQTT with SSL via GUI and check the logs

### Screenshots
Config:
```
{
  "embedded": true,
  "robot": {
    "implementation": "auto",
    "implementationSpecificConfig": {
      "ip": "127.0.0.1"
    }
  },
  "webserver": {
    "port": 80,
    "basicAuth": {
      "enabled": true,
      "username": "XXX",
      "password": "XXX"
    }
  },
  "zonePresets": {},
  "goToLocationPresets": {},
  "mqtt": {
    "enabled": true,
    "server": "mqtthost.myinternal.domain",
    "port": 8883,
    "username": "XXX",
    "password": "XXX",
    "ca": "-----BEGIN CERTIFICATE-----\nXXX\n-----END CERTIFICATE-----",
    "clientCert": "",
    "clientKey": "",
    "identifier": "XXX",
    "friendlyName": "XXX",
    "homie": {
      "enabled": false,
      "addICBINVMapProperty": false,
      "cleanAttributesOnShutdown": false
    },
    "homeassistant": {
      "enabled": true,
      "cleanAutoconfOnShutdown": false
    },
    "provideMapData": true
  },
  "ntpClient": {
    "enabled": true,
    "server": "ntphost.myinternal.domain",
    "port": 123,
    "interval": 28800000,
    "timeout": 10000
  },
  "timers": {},
  "allowSSHKeyUpload": true,
  "logLevel": "info",
  "debug": {
    "memoryStatInterval": false,
    "enableDebugCapability": false,
    "debugHassAnchors": false
  },
  "networkAdvertisement": {
    "enabled": true
  }
}
```

Valetudo error:
```
[2021-08-31T12:40:48.805Z] [ERROR] MQTT error: Error: read ECONNRESET
```

MQTT server error:
```
1630413665: Client connection from 192.168.6.3 failed: error:1408F10B:SSL routines:ssl3_get_record:wrong version number.
```

### Vacuum Model
rockrobo.vacuum.v1

### Valetudo Version
2021.08.1

## Expected behavior
MQTT with SSL working as per 2021.08.0

Hypfer, 2021-08-31:
Fixed in f00edb32767a01ed71455fab13b57c205cf4dae2

Note that that is a breaking change and might require reconfiguration of valetudo

closed 2021-08-31 by Hypfer

darootler, 2021-08-31:
Thank you.

Regards
Richard