Hypfer/Valetudo #1065: vendor.dreame: Manual Control

hereticmaggot, 2021-09-02:
Hello, I see in the autogenerated changelog of 2021.08.0 release that vendor.dreame: Manual Control was added.
However, I don't see how to use it. Is there a way using the rest api?

ingoschi, 2021-09-02:
see /swagger/#/ManualControlCapability

hereticmaggot, 2021-09-03:
Thank you I see the commands. However applying them does not work.
The robot indeed says starting remote cleanup but never moves.

ingoschi, 2021-09-03:
I haven't tried it myself yet.
But according to https://gist.github.com/Hypfer/fcfa39996bd7522bbe2c5f18acb1fcf4#file-valetudo_gamepad_experiments_v1-js
you first have to send 

{"action":"enable"}

 to /api/v2/robot/capabilities/ManualControlCapability
before you send a.e.

 {"action": "move",
   "movementCommand": "forward" }
 
And I guess  {"action":"disable"} when you're finished with your movements.

hereticmaggot, 2021-09-03:
Thanks it works.
Just would be nice to have that function integrated into Valetudo UI

PR #1068 (2021-09-05): Swagger examples for ManualControlCapability
## Type of change

Type A:
- [ ] Bug fix (non-breaking change which fixes an issue)
- [ ] Refactor/Code Cleanup
- [ x] Docs
- [ ] Capability implementation for existing core capability
- [ ] New robot implementation
  
Type B:
- [ ] New capability
- [ ] New core feature

# Description (Type A)
ManualControlCapability had now swagger examples, the generated example value was wrong
 
Fixes #987
Fixes #1065
closed 2021-09-03 by hereticmaggot