Hypfer/Valetudo #1067: Mqtt topic for hass not correct after upgrade, sets to "undefined"

pkishino, 2021-09-03:
## Describe the bug

Before the upgrade the mqtt topic for hass integration was the following: valetudo>robot>hass
Now, when checking mqtt it is sending to undefined>robot>hass
[image]

the other parts under valetudo>robot update correctly

## To Reproduce

I have tried to restart mqtt/restart vacuum/enable-disable hass integration etc.. 
looking at the code it seem the topic is never set for this.

### Screenshots

### Vacuum Model

Xiaomi gen 1

### Valetudo Version

2021.8.1
## Expected behavior

the hass integration/autodiscovery should write to valetudo>robot>hass topic

## Additional context

Hypfer, 2021-09-05:
Indeed :(

When working on 65a335ba, I've missed that the HassController also reads the config file by itself which led to this.
This wasn't discovered during testing as home assistant just started using the topics starting with `undefined/`

f00edb32 fixes this. Note that that is a breaking change and will require reconfiguration due to the different MQTT config schema

closed 2021-09-05 by Hypfer

pkishino, 2021-09-05:
Awesome, thanks for the fix, looking forward to this soon. Yeah, it works but just looks strange, thanks!