Hypfer/Valetudo #1076: ViomiV7ValetudoRobot ---- reason: Error: No sockets available, cannot start.

johnappletree, 2021-09-10:
### Vacuum Model
Robot: Viomi Xiaomi Mi Robot Vacuum-Mop P (ViomiV7ValetudoRobot)

### Valetudo Version
2021.09.0

Hi there,

I lost my robot after upgrading to 2021.08.1. I connected using adb to extract the valetudo logs, They look like this:

`[1970-01-01T00:20:23.942Z] [INFO] Set Logfile to /tmp/valetudo.log
[1970-01-01T00:20:23.961Z] [INFO] Autodetected ViomiV7ValetudoRobot
[1970-01-01T00:20:24.070Z] [INFO] Starting Valetudo 2021.09.0
[1970-01-01T00:20:24.073Z] [INFO] Commit ID: 8874f7b9203fe480e22e2e76e6ba4f96230e2d28
[1970-01-01T00:20:24.077Z] [INFO] Configuration file: /mnt/UDISK/valetudo_config.json
[1970-01-01T00:20:24.078Z] [INFO] Logfile: /tmp/valetudo.log
[1970-01-01T00:20:24.079Z] [INFO] Robot: Viomi Xiaomi Mi Robot Vacuum-Mop P (ViomiV7ValetudoRobot)
[1970-01-01T00:20:24.080Z] [INFO] JS Runtime Version: v16.4.1
[1970-01-01T00:20:24.082Z] [INFO] Arch: arm
[1970-01-01T00:20:24.084Z] [INFO] Max Heap Size: 34 MiB
[1970-01-01T00:20:24.085Z] [INFO] Node Flags: --expose-gc --max-heap-size=34 --sparkplug
[1970-01-01T00:20:24.098Z] [INFO] Autogenerated System ID: FairYouthfulMandrill
[1970-01-01T00:20:24.112Z] [INFO] DeviceId 375360940
[1970-01-01T00:20:24.113Z] [INFO] IP 127.0.0.1
[1970-01-01T00:20:24.114Z] [INFO] CloudSecret 
[1970-01-01T00:20:24.116Z] [INFO] LocalSecret 
[1970-01-01T00:20:32.882Z] [INFO] Valetudo can be reached via: valetudo_fairyouthfulmandrill.local
[1970-01-01T00:20:32.910Z] [INFO] Setting /proc/self/oom_score_adj to 666. Previous value: -1000
[1970-01-01T00:20:32.925Z] [INFO] Dummycloud is spoofing 127.0.0.1:8053 on 127.0.0.1:8053
[1970-01-01T00:20:32.937Z] [INFO] Webserver running on port 80
[1970-01-01T00:20:32.954Z] [INFO] Map Upload Server running on port 8079
[1970-01-01T00:20:32.977Z] [INFO] Bonjour service "Valetudo Xiaomi Mi Robot Vacuum-Mop P Web" with type "http" started
[1970-01-01T00:20:32.978Z] [INFO] Bonjour service "Valetudo Xiaomi Mi Robot Vacuum-Mop P" with type "valetudo" started
[1970-01-01T00:20:32.989Z] [ERROR] unhandledRejection {
  reason: Error: No sockets available, cannot start.
      at SsdpServer.SSDP._createSockets (/snapshot/Valetudo/node_modules/node-ssdp/lib/index.js:186:11)
      at SsdpServer.SSDP._start (/snapshot/Valetudo/node_modules/node-ssdp/lib/index.js:229:10)
      at /snapshot/Valetudo/node_modules/node-ssdp/lib/server.js:79:10
      at Promise._execute (/snapshot/Valetudo/node_modules/bluebird/js/release/debuggability.js:384:9)
      at Promise._resolveFromExecutor (/snapshot/Valetudo/node_modules/bluebird/js/release/promise.js:518:18)
      at new Promise (/snapshot/Valetudo/node_modules/bluebird/js/release/promise.js:103:10)
      at SsdpServer.start (/snapshot/Valetudo/node_modules/node-ssdp/lib/server.js:72:10)
      at NetworkAdvertisementManager.setUpSSDP (/snapshot/Valetudo/backend/lib/NetworkAdvertisementManager.js:51:25)
      at NetworkAdvertisementManager.setUp (/snapshot/Valetudo/backend/lib/NetworkAdvertisementManager.js:32:18)
      at new NetworkAdvertisementManager (/snapshot/Valetudo/backend/lib/NetworkAdvertisementManager.js:22:14)
      at new Valetudo (/snapshot/Valetudo/backend/lib/Valetudo.js:84:44)
      at Object.<anonymous> (/snapshot/Valetudo/backend/index.js:6:16)
      at Module._compile (node:internal/modules/cjs/loader:1095:14)
      at Module._compile (pkg/prelude/bootstrap.js:1758:32)
      at Object.Module._extensions..js (node:internal/modules/cjs/loader:1124:10)
      at Module.load (node:internal/modules/cjs/loader:975:32),
  stack: 'Error: No sockets available, cannot start.\n' +
    '    at SsdpServer.SSDP._createSockets (/snapshot/Valetudo/node_modules/node-ssdp/lib/index.js:186:11)\n' +
    '    at SsdpServer.SSDP._start (/snapshot/Valetudo/node_modules/node-ssdp/lib/index.js:229:10)\n' +
    '    at /snapshot/Valetudo/node_modules/node-ssdp/lib/server.js:79:10\n' +
    '    at Promise._execute (/snapshot/Valetudo/node_modules/bluebird/js/release/debuggability.js:384:9)\n' +
    '    at Promise._resolveFromExecutor (/snapshot/Valetudo/node_modules/bluebird/js/release/promise.js:518:18)\n' +
    '    at new Promise (/snapshot/Valetudo/node_modules/bluebird/js/release/promise.js:103:10)\n' +
    '    at SsdpServer.start (/snapshot/Valetudo/node_modules/node-ssdp/lib/server.js:72:10)\n' +
    '    at NetworkAdvertisementManager.setUpSSDP (/snapshot/Valetudo/backend/lib/NetworkAdvertisementManager.js:51:25)\n' +
    '    at NetworkAdvertisementManager.setUp (/snapshot/Valetudo/backend/lib/NetworkAdvertisementManager.js:32:18)\n' +
    '    at new NetworkAdvertisementManager (/snapshot/Valetudo/backend/lib/NetworkAdvertisementManager.js:22:14)\n' +
    '    at new Valetudo (/snapshot/Valetudo/backend/lib/Valetudo.js:84:44)\n' +
    '    at Object.<anonymous> (/snapshot/Valetudo/backend/index.js:6:16)\n' +
    '    at Module._compile (node:internal/modules/cjs/loader:1095:14)\n' +
    '    at Module._compile (pkg/prelude/bootstrap.js:1758:32)\n' +
    '    at Object.Module._extensions..js (node:internal/modules/cjs/loader:1124:10)\n' +
    '    at Module.load (node:internal/modules/cjs/loader:975:32)',
  promise: Promise [Object] {
    _bitField: 18087936,
    _fulfillmentHandler0: Error: No sockets available, cannot start.
        at SsdpServer.SSDP._createSockets (/snapshot/Valetudo/node_modules/node-ssdp/lib/index.js:186:11)
        at SsdpServer.SSDP._start (/snapshot/Valetudo/node_modules/node-ssdp/lib/index.js:229:10)
        at /snapshot/Valetudo/node_modules/node-ssdp/lib/server.js:79:10
        at Promise._execute (/snapshot/Valetudo/node_modules/bluebird/js/release/debuggability.js:384:9)
        at Promise._resolveFromExecutor (/snapshot/Valetudo/node_modules/bluebird/js/release/promise.js:518:18)
        at new Promise (/snapshot/Valetudo/node_modules/bluebird/js/release/promise.js:103:10)
        at SsdpServer.start (/snapshot/Valetudo/node_modules/node-ssdp/lib/server.js:72:10)
        at NetworkAdvertisementManager.setUpSSDP (/snapshot/Valetudo/backend/lib/NetworkAdvertisementManager.js:51:25)
        at NetworkAdvertisementManager.setUp (/snapshot/Valetudo/backend/lib/NetworkAdvertisementManager.js:32:18)
        at new NetworkAdvertisementManager (/snapshot/Valetudo/backend/lib/NetworkAdvertisementManager.js:22:14)
        at new Valetudo (/snapshot/Valetudo/backend/lib/Valetudo.js:84:44)
        at Object.<anonymous> (/snapshot/Valetudo/backend/index.js:6:16)
        at Module._compile (node:internal/modules/cjs/loader:1095:14)
        at Module._compile (pkg/prelude/bootstrap.js:1758:32)
        at Object.Module._extensions..js (node:internal/modules/cjs/loader:1124:10)
        at Module.load (node:internal/modules/cjs/loader:975:32),
    _rejectionHandler0: undefined,
    _promise0: undefined,
    _receiver0: undefined
  }
}
[1970-01-01T00:20:33.033Z] [INFO] Got token from handshake: 4935566647637243355374446f656b55
[1970-01-01T00:20:34.112Z] [WARN] Error during time sync: Error: Timeout waiting for NTP response from pool.ntp.org:123
`

I flashed 2021.09.0 using ADB hoping that the bug might be fixed. But the problem still exists. Is this a known bug? 
I do not understand why this code refers to /snapshot/Valetudo/ This directory does not exist on my device.

closed 2021-09-12 by Hypfer

johnappletree, 2021-09-12:
Dear Hypfer, could you please explain why this issue is invalid? Is there some information missing in the opening post? I am happy to provide more information if needed.

Hypfer, 2021-09-13:
The error you're seeing there isn't crashing the process and therefore can't be responsible for

"I lost my robot"

whatever that might mean exactly.
It does seem to appear sometimes when Valetudo is started multiple times. Probably something broken in the ssdp library. Valetudo still works even if that happens though.

This likely is a deployment issue (as it always is on Viomi) and therefore not a bug but a support request.