Hypfer/Valetudo #1111: Viomi. Error while adjusting timezone to UTC

DuBass858, 2021-09-27:
## Describe the bug

All the time the robot cannot set the correct time zone, therefore the time is not correct.

## To Reproduce

After reboot you can see all the time

### Screenshots

### Vacuum Model

ViomiV7
### Valetudo Version

2021.09.1
## Expected behavior

The robot must set the correct time

## Additional context

[2021-07-28T19:24:26.339Z] [DEBUG] local request 5009 set_timezone timed out
[2021-07-28T19:24:26.343Z] [DEBUG] >>> local: {"method":"set_timezone","params":[0],"id":7011}
[2021-07-28T19:24:26.908Z] [DEBUG] local request 6010 set_timezone timed out
[2021-07-28T19:24:26.918Z] [DEBUG] >>> local: {"method":"set_timezone","params":[0],"id":8012}
[2021-09-27T18:40:46.002Z] [INFO] Successfully set the robot time via NTP to 2021-09-27T18:40:46.894Z
[2021-09-27T18:40:48.444Z] [WARN] Token is okay, however we're unable to reach the vacuum { retries: 10, method: 'set_timezone', args: [ 0 ] }
[2021-07-28T19:24:40.509Z] [INFO] Cloud message timed out. Assuming that we're not connected anymore
[2021-07-28T19:24:40.511Z] [ERROR] RobotMQTTHandle: Error while polling the robots state MiioTimeoutError: request timed out:{"method":"get_prop","params":["run_state","mode","err_state","battary_life","box_type","mop_type","s_time","s_area","suction_grade","water_grade","remember_map","has_map","is_mop","has_newmap","timezone"],"id":13}
    at Timeout.timeout [as _onTimeout] (/snapshot/Valetudo/backend/lib/miio/MiioSocket.js:210:32)
    at listOnTimeout (node:internal/timers:557:17)
    at processTimers (node:internal/timers:500:7)
[2021-07-28T19:24:44.017Z] [INFO] Cloud connected
[2021-07-28T19:25:12.786Z] [WARN] Error while adjusting timezone to UTC

DuBass858, 2021-10-02:
Valetudo Version
2021.10.0

[2021-07-28T19:24:45.674Z] [DEBUG] Starting NTP Query for { server: 'pool.ntp.org', port: 123, timeout: 10000 }
[2021-07-28T19:24:45.733Z] [DEBUG] Got Time from NTP Server: 2021-10-02T09:41:53.894Z
[2021-10-02T09:41:53.002Z] [INFO] Successfully set the robot time via NTP to 2021-10-02T09:41:53.894Z
[2021-10-02T09:41:53.005Z] [DEBUG] Next NTP sync in 28800000 ms
[2021-07-28T19:24:48.704Z] [DEBUG] local request 10016 set_timezone timed out
[2021-07-28T19:24:48.713Z] [DEBUG] >>> local: {"method":"set_timezone","params":[0],"id":13019}
[2021-07-28T19:24:49.826Z] [DEBUG] >>> cloud: {"method":"set_timezone","params":[0],"id":8}
[2021-07-28T19:24:50.166Z] [DEBUG] GC forced at 51634176 bytes RSS freed 1654784 bytes of memory.
[2021-07-28T19:24:50.169Z] [DEBUG] local request 11017 set_timezone timed out
[2021-07-28T19:24:50.175Z] [DEBUG] >>> local: {"method":"set_timezone","params":[0],"id":14020}
[2021-07-28T19:24:50.944Z] [DEBUG] local request 12018 set_timezone timed out
[2021-07-28T19:24:50.947Z] [DEBUG] >>> local: {"method":"set_timezone","params":[0],"id":15021}

Hypfer, 2021-12-27:
https://github.com/Hypfer/Valetudo/issues/1156#issuecomment-1001695151

closed 2021-12-27 by Hypfer

nekromant, 2021-12-28:
@Hypfer Thanks, deploying these hacks to my robots!