Hypfer/Valetudo #1126: Vacuum state is not consistent

ibanmda, 2021-10-03:
## Describe the bug

Vacuum state is not consistent and fails on Dreame 1c

## To Reproduce

Steps to reproduce the behavior (states):
1. Changed: Returning to the dock
2. Changed: On the dock
3. Changed: Cleaning
4. Changed: On the dock

### Screenshots
[image]

### Vacuum Model
Dreame 1c

### Valetudo Version
2021.09.1

## Expected behavior
When the vacuum finishes cleaning, returns to the dock, it goes to the dock, and remains at the dock (not to Cleaning again).

## Additional context

Meaxl, 2021-10-03:
Can confirm that behviour @ibanmda describes.

Hope for a fix soon, as of this misbehavior for example I'm currently not able to calculate the time for the last cleaning cycle. Using MQTT Triggers to start at cleaning and stop on state docked.

dugite-code, 2021-10-12:
@ibanmda @Meaxl Do you have the `Delete autodiscovery on shutdown` enabled in the MQTT settings? I think turning that off has improved the situation for me (at least initially we'll see long term might have just refreshed the auto discover topics)

Hypfer, 2021-11-20:
c73e0620 should fix this. if not please reopen

closed 2021-11-20 by Hypfer