Hypfer/Valetudo #1135: New Valetudo.cloud Template/Theme

Hypfer, 2021-10-06:
Currently, https://valetudo.cloud uses the most basic theme that was available at the time.
While it sorta works, it's also a low hanging fruit providing much potential for perceived improvement with relatively little required effort.

Requirements:
- Should work well on both mobile as well as desktops
- No externally hosted resources

and that's pretty much it.

The site uses github pages and therefore jekyll.
It can be found here: https://github.com/Hypfer/Valetudo/tree/master/docs

Local testing can be done by running `bundle exec jekyll serve`.

Hypfer, 2021-10-06:
Fancy!

We definitely need stable URLs though as I'm pretty sure that they're used in multiple places where we can't change them

joaopgrassi, 2021-12-09:
I moved my blog from Ghost to Hugo hosted on GH pages and you can pretty much fix the urls by setting it up in frontmatter. If it helps, you can see one old post in my blog that I had to maintain the url https://raw.githubusercontent.com/joaopgrassi/blog/main/src/content/posts/2019/typed-httpclient-identityserver/index.md

You just need to set baseurl and all will work https://github.com/joaopgrassi/blog/blob/a2231c5e90437470fdac6a4b622fbe96a1999ed0/src/config.yml#L1

Mayank1728, 2022-02-15:
hey @Hypfer  !  I would like to contribute however I have quite basic knowledge of Js

closed 2022-11-29 by Hypfer