Hypfer/Valetudo #1147: customOrder Segment Cleanup broken on Z10 Pro

ccoors, 2021-10-11:
## Describe the bug

The Dreame Z10 Pro does not respect the custom cleanup order for segmented cleanups. 

## To Reproduce

1. Start a segmented cleanup with `customOrder: true` and an unusual segment order via REST/MQTT
2. Observe the robot simply cleaning the "nearest" segment first (or some other metric? But that's what I observed)

### Screenshots

#### Segments
[image]

#### Cleanup command with customOrder
[image]

#### Result
Still cleans Foyer before Kitchen
[image]

### Vacuum Model
Dreame Z10 Pro on FW 1056

### Valetudo Version
2021.10.0

## Expected behavior

Either Valetudo should not report `customOrderSupport: true` or the cleanup should be performed in the specified order.

## Additional context

This did work on my D9 on FW 1093. Maybe the command is different or just unsupported? A first look at the code didn't reveal an obvious bug.

pkoehlers, 2022-01-06:
I just double checked this with my Dreame L10 Pro on latest firmware 1105. It is working as intended in my setup. 

Actually I also checked the code and the `customOrderSupport` is not evaluated for Dreame robots. I would say that my robot just uses the order that I passed the segment ids in. Order just works when sending this via MQTT: `{"segment_ids": ["7", "11", "9", "1"]}`

 I agree that it is misleading that Swagger's default request includes this parameter. But if you check the schema tab in the Swagger UI, you will see that `customOrderSupport` is commented as "(Needs firmware support)" and it shouldn't do harm sending it.

Hypfer, 2022-04-18:
I've been using custom order segment cleanups with FW 1099 for a while now.

closed 2022-04-18 by Hypfer