Hypfer/Valetudo #1165: Cleaning seems to crash with Dreame D9

eikel, 2021-10-16:
## Describe the bug

When I start the cleaning using the Valetudo web UI, the robot behaves weird. Sometimes it starts moving, and then suddenly stops without doing anything. It is then not reachable on the network anymore until I put it back onto the docking station. The battery is reported to be at 100%.  Sometimes the robot does not start to move and the robot's voice says something like “left wheel is blocked”, “main brush is blocked”, or “front bumber is blocked”. I checked the hardware, but could not detect any problems. I had the case once that the error "UNKNOWN ERROR CODE 31,32" was published to the web UI and MQTT. As these symptoms are rather random, I assume the problem is not really connected to these errors. I am unsure how to proceed investigating this problem. Are there additional log files available that I could check?

## To Reproduce

1. Click on “Start” in the Valetudo web UI.
2. Wait.

### Screenshots

Here is my log from `/tmp/valetudo.log`:
```
[1970-01-01T00:10:07.914Z] [INFO] Set Logfile to /tmp/valetudo.log
[1970-01-01T00:10:07.940Z] [INFO] Autodetected DreameD9ValetudoRobot
[1970-01-01T00:10:08.077Z] [INFO] Starting Valetudo 2021.10.0
[1970-01-01T00:10:08.080Z] [INFO] Commit ID: 01088ecdadaaefbb7e2b2d812d2377a01277bf00
[1970-01-01T00:10:08.081Z] [INFO] Configuration file: /data/valetudo_config.json
[1970-01-01T00:10:08.081Z] [INFO] Logfile: /tmp/valetudo.log
[1970-01-01T00:10:08.082Z] [INFO] Robot: Dreame D9 (DreameD9ValetudoRobot)
[1970-01-01T00:10:08.083Z] [INFO] JS Runtime Version: v16.10.0
[1970-01-01T00:10:08.083Z] [INFO] Arch: arm
[1970-01-01T00:10:08.084Z] [INFO] Max Heap Size: 34 MiB
[1970-01-01T00:10:08.085Z] [INFO] Node Flags: --expose-gc --max-heap-size=34 --optimize-for-size --lite-mode --no-regexp-tier-up --no-expose-wasm
[1970-01-01T00:10:08.098Z] [INFO] Autogenerated System ID: NeighboringPriceyOpossum
[1970-01-01T00:10:08.112Z] [INFO] DeviceId *censored*
[1970-01-01T00:10:08.113Z] [INFO] IP 127.0.0.1
[1970-01-01T00:10:08.113Z] [INFO] CloudSecret *censored*
[1970-01-01T00:10:08.114Z] [INFO] LocalSecret *censored*
[1970-01-01T00:10:08.116Z] [INFO] Firmware Version: 1093
[1970-01-01T00:10:08.116Z] [INFO] MCU Version: 0.8.6
[1970-01-01T00:10:08.120Z] [INFO] Current rootfs: rootfs2 (nand0p6)
[1970-01-01T00:10:25.074Z] [INFO] Valetudo can be reached via: valetudo_neighboringpriceyopossum.local
[1970-01-01T00:10:25.102Z] [INFO] Setting /proc/self/oom_score_adj to 666. Previous value: 0
[1970-01-01T00:10:25.117Z] [INFO] Dummycloud is spoofing 127.0.0.1:8053 on 127.0.0.1:8053
[1970-01-01T00:10:25.120Z] [INFO] Webserver running on port 80
[1970-01-01T00:10:25.137Z] [INFO] SSDP/UPnP advertisement started
[1970-01-01T00:10:25.149Z] [INFO] Map Upload Server running on port 8079
[1970-01-01T00:10:25.167Z] [INFO] Bonjour service "Valetudo D9 Web" with type "http" started
[1970-01-01T00:10:25.168Z] [INFO] Bonjour service "Valetudo D9" with type "valetudo" started
[1970-01-01T00:10:25.177Z] [WARN] Error during time sync: Error: Timeout waiting for NTP response from fritz.box:123
    at Object.getNetworkTime (/snapshot/Valetudo/node_modules/@destinationstransfers/ntp/index.js:64:11)
    at async NTPClient.pollTime (/snapshot/Valetudo/backend/lib/NTPClient.js:56:36)
[1970-01-01T00:10:25.614Z] [ERROR] MQTT error: Error: certificate is not yet valid
[1970-01-01T00:10:26.808Z] [ERROR] MQTT error: Error: certificate is not yet valid
[1970-01-01T00:10:28.139Z] [ERROR] MQTT error: Error: certificate is not yet valid
[1970-01-01T00:10:29.299Z] [ERROR] MQTT error: Error: certificate is not yet valid
[1970-01-01T00:10:30.121Z] [INFO] Cloud connected
[1970-01-01T00:10:30.689Z] [ERROR] MQTT error: Error: certificate is not yet valid
[1970-01-01T00:10:31.887Z] [ERROR] MQTT error: Error: certificate is not yet valid
[…]
[1970-01-01T00:11:23.306Z] [ERROR] MQTT error: Error: certificate is not yet valid
[1970-01-01T00:11:24.491Z] [ERROR] MQTT error: Error: certificate is not yet valid
[2021-10-16T17:26:33.003Z] [INFO] Successfully set the robot time via NTP to 2021-10-16T17:26:33.894Z
[2021-10-16T17:26:33.655Z] [INFO] Connected successfully to MQTT broker
[2021-10-16T17:26:33.847Z] [INFO] MQTT configured
[2021-10-16T17:26:35.904Z] [INFO] Cloud message timed out. Assuming that we're not connected anymore
[2021-10-16T17:26:35.907Z] [ERROR] RobotMQTTHandle: Error while polling the robots state MiioTimeoutError: request timed out:{"method":"get_properties","params":[{"siid":4,"piid":1,"did":391898788},{"siid":4,"piid":7,"did":391898788},{"siid":4,"piid":4,"did":391898788},{"siid":4,"piid
":5,"did":391898788},{"siid":4,"piid":6,"did":391898788},{"siid":4,"piid":18,"did":391898788},{"siid":3,"piid":1,"did":391898788},{"siid":3,"piid":2,"did":391898788}],"id":2}
    at Timeout.timeout [as _onTimeout] (/snapshot/Valetudo/backend/lib/miio/MiioSocket.js:210:32)
    at listOnTimeout (node:internal/timers:557:17)
    at processTimers (node:internal/timers:500:7)
[2021-10-16T17:26:36.184Z] [INFO] Cloud connected
```
I started the cleaning after the last log line appeared, so the problem situation is unfortunately not visible in this log.

### Vacuum Model

Dreame D9

### Valetudo Version

2021.10.0 lowmem

## Expected behavior

Cleaning works 

## Additional context

I have flashed the robot with an image from DustBuilder (firmware “D9 (ver 1093, ??/2021)”) using the “WiFi reset UART root” method. I checked the files that are mentioned in the FAQ to be backed up, but those still have the same contents as they had before flashing.
Before rooting the device, cleaning worked without problems for several weeks.

Hypfer, 2021-10-16:
Did you look into the system resources?

That sounds like their main control process (ava) dying pretty hard.

Hypfer, 2021-10-16:
`VmPeak:   236296 kB`

That's 100% of available system memory.
=> Sounds like a catastrophic memory leak in ava.

On my Z10 these values look like this:

```
VmPeak:  1283220 kB
VmSize:  1280960 kB
```