Hypfer/Valetudo #1167: "Water Tank Is Not Installed" message when trying to vacuum

alex3s, 2021-10-17:
## Describe the bug
The vacuum doesn't want to start when the dustbin is in (the most often used accessory).
"Water Tank Is Not Installed" message appears.
Vacuum is blinking red.

## To Reproduce
Remove the 2in1 water tank dustbin and place the higher capacity dust bin which doesn't have water compartment.
Press START on Valetudo home screen.

### Vacuum Model
Viomi 2 Pro, running ViomiV7ValetudoRobot

### Valetudo Version
2021.10.0 armv7

## Expected behavior
Should start the vacuum run when pressed start button on the Valetudo screen.

## Additional context
It starts correctly when pressing the hardware button on the device or when 
`miiocli viomivacuum --ip 192.168.100.100 --token xxxxxxx start` invoked from Linux console
It also starts correctly when the 2in1 watertank dustbin is placed in instead of simply dustbin.
No other known ways to start vacuuming from home screen, however the older version used to run correctly (2 versions before if my memory doesn't fail me)

drticcz, 2022-08-15:
This just suddenly happened to me after upgrade to the 2022.08 version but I doubt the upgrade is actually the origin. It has been vacuuming for several cycles just fine after the upgrade.

My symptoms are worse. When I insert the mopping dustbin, it says "watertank has been inserted" but then won't start with an error "Dust bin or 2-in-1 dust bin is not installed"

drticcz, 2022-08-17:
So as expected, this has nothing to do with Valetudo or its version... When you install the HEPA filter to one of your tanks / dust bins, check carefully if that filter has the NFC chip on it. I manage to lose one from the filter during the filter cleanup. It looks like as mall black pinhead - see the picture below, left side with the nfc chip, right side with the chip missing. It won't work if you use the filter without chip....

[image]

Blackchickenn, 2022-09-05:
This (or something similar) happened to me aswell couple days ago. I am using still dust tank and when I start the device it says something like "install dust tank or 2in1 water tank". I upgraded firmware, reset wifi, reset device to factory reset. Aswell I checked the black pin mentioned above on both and there are there.  When I change dust tank for 2in1 water tank it says 2in1 was installed but after start same error "install dust tank or 2in1 water tank" . Any ideas what could be wrong?

Hypfer, 2022-11-29:
> Vacuum is blinking red.

With the vacuum blinking read, it's a firmware thing and not a Valetudo thing.
Fortunately, @drticcz already figured out what the real issue is. Thank you

closed 2022-11-29 by Hypfer