Hypfer/Valetudo #1194: The lab status is the same

xuefer, 2021-11-05:
## Describe the bug
"The lab status is the same" error when enabling persistant data
## To Reproduce
1. Robot Settings
2. Persistant data was off, enable it

### Screenshots
missed the chance to create screenshot, i don't want to turn it off to reproduce it. since it will delete my current map

### Vacuum Model
Roborock T6 (S6)

### Valetudo Version
2021.11.0

## Expected behavior
Show as "on" after i switch it on

## Additional context
It seems after first time I switch it on, no error reported. Its actual value is on but not shown as on. subsequence switching it on raise the error. refresh the page get it show as on as expected

xuefer, 2021-11-05:
```
[2021-11-05T03:16:22.050Z] [INFO] Miio error response {
  id: 4153,
  error: { code: -10005, message: 'The Lab Status is same' },
  exe_time: 1
}
```

Hypfer, 2021-11-05:
Problem is that no one ever bothered to research how roborock robots newer than the s5 do their map handling which is why this is more or less broken on everything that isn't an S5

closed 2021-11-07 by Hypfer