Hypfer/Valetudo #1210: Roborock disoriented after first run with NoGo Area

podkilla, 2021-11-13:
## Describe the bug
Robot becomes disoriented after not being able to circle nogo area. It stays disoriented for future runs until map is reset and a normal run regenerated the map.

## To Reproduce

Steps to reproduce the behavior:
1. Fresh installation
2. First run to get map (persistent Map enabled)
3. Create NoGo Area in WebIF
4. Let the vaccuum clean segment with NoGo area. During this run it is unable to circle the nogo area because of chair legs. It then starts navigating in circles. 
5. All the next runs look like the picture below - i canceled it though.

### Screenshots

### Vacuum Model

Roborock S5
### Valetudo Version

2021.11.0
## Expected behavior

## Additional context

Hypfer, 2021-11-13:
This is neither a bug in valetudo nor has it anything to do with the no go area.

The side sensor is dirty

closed 2021-11-13 by Hypfer

podkilla, 2021-11-13:
Wrong - i cleaned everything multiple times. This has nothing to do with any sensor. The robot runs perfectly fine without the nogo area for weeks.

rimrul, 2021-11-15:
Still, not a bug in Valetudo. The robot firmware does all the pathing logic.

minils, 2022-01-30:
I'm facing a similar issue where the robot gets confused and ends its cleaning early after it cleans for the first time. It starts its way as always following the edges of the rooms but after cleaning a small part returns to its dock through very crazy paths, skipping the rest of our apartment.  Will upload a screenshot later.