Hypfer/Valetudo #1226: ntpClient.server is ignored in config file

xuefer, 2021-11-21:
this bug was reported in #716 already but was closed

## Describe the bug
valetudo uses pool.ntp.org regardless of settings in /mnt/data/valetudo_config.json

## To Reproduce
1. Set ntpClient.server to something else in /mnt/data/valetudo_config.json
2. start /usr/local/bin/valetudo manually

### Screenshots
```
[2021-11-21T15:32:15.402Z] [WARN] Error during time sync: Error: Timeout waiting for NTP response from pool.ntp.org:123
    at Object.getNetworkTime (/snapshot/Valetudo/node_modules/@destinationstransfers/ntp/index.js:64:11)
    at async NTPClient.pollTime (/snapshot/Valetudo/backend/lib/NTPClient.js:56:36)
[2021-11-21T15:33:16.006Z] [INFO] Successfully set the robot time via NTP to 2021-11-21T15:33:16.895Z
```

### Vacuum Model
Roborock T6

### Valetudo Version
2021.11.1

## Expected behavior
use specified ntp server

## Additional context

Hypfer, 2021-11-21:
I'm pretty sure that it does.

[image]

Did you edit the config file by hand? If it doesn't pass schema validation, valetudo will replace it with the defaults.

xuefer, 2021-11-21:
actually it was configured correctly in #/settings/connectivity

[image]

closed 2021-11-21 by xuefer

Hypfer, 2021-11-21:
Updating it via the UI does work and it also updates the config.

Were you looking at an older log entry?

xuefer, 2021-11-21:
the problem was i fogot to run "export VALETUDO_CONFIG_PATH=/mnt/data/valetudo_config.json", never mind. but i'm looking at why it failed at first attempt

```
[2021-11-21T16:08:22.837Z] [WARN] Error during time sync: Error: Timeout waiting for NTP response from router.lan:123
    at Object.getNetworkTime (/snapshot/Valetudo/node_modules/@destinationstransfers/ntp/index.js:64:11)
    at async NTPClient.pollTime (/snapshot/Valetudo/backend/lib/NTPClient.js:56:36)
[2021-11-21T16:08:22.956Z] [INFO] Connected successfully to MQTT broker
```
tcpdump on ntp server see nothing during this error

but subsequence retry success output "Successfully set the robot time via NTP to 2021-11-21T16:09:23.895Z" and tcpdump can see the request and reponse are sent during this success log