Hypfer/Valetudo #1232: New frontend crashes on iOS

jay187, 2021-11-23:
When opening the new frontend either with Safari or Chrome under iOS 15.1 the browser crashes after a reload of the page.
I've also cleared the browser cache and website data but this didn't help either.
Using the old frontend still works though.

## To Reproduce

### Screenshots

[image]

### Vacuum Model

Dreame D9

### Valetudo Version

2021.11.1

## Expected behavior

## Additional context

Hypfer, 2021-11-23:
Unfortunately due to Apple being Apple, there is no way of debugging this without spending thousands of dollars for apple-certified hardware.
The code itself is valid and works fine on all platforms that aren't Apple so it's not a bug in Valetudo.

Anyways, I can't do anything about that so I'll leave this open for someone willing to pay the Apple tax to figure out.

Hypfer, 2021-11-23:
Sidenote:

The only map-related change between 2021.11.0 and 2021.11.1 was that there can now be multiple path entities on dreame resulting in multiple svgs being drawn onto the map canvas.
If someone decides to pick this up that might be a good place to start looking.

jay187, 2021-11-23:
@Hypfer Thanks for the quick reply. Just some more context: It seems like an iOS 15 problem. On my other iPhone running iOS 14 the frontend works as expected.

maciejtarmas, 2021-11-24:
Can confirm. Looks like more complex maps crash Safari on iOS. Same map on Safari on macOS causes it to display “This page is slowing down Safari” message. Seems that some maps cause a resource usage spike when being drawn.

Source: my 2 Dreame D9 robots. One works, and one crashes Safari. Same map works fine on Firefox though.

Haven’t really started looking into it, so I haven’t filed a bug report myself.

Hypfer, 2021-11-24:
@maciejtarmas 
> Same map works fine on Firefox though.

Firefox on iOS? That is _very_ strange as all browsers on iOS are using the Safari Webkit implementation as their rendering framework meaning that there only is Safari on iOS.

Why would it crash Safari Safari but not Firefox Safari?

See: https://en.wikipedia.org/wiki/Firefox_for_iOS
>2.5.6 Apps that browse the web must use the appropriate WebKit framework and WebKit Javascript.

jay187, 2021-11-24:
@Hypfer It's working now. Did not change anything on the phone but I did vaccum today. Maybe due to that and the fact the map has new data it's working now? So this might be a only an issue after the upgrade when no run was made?

Hypfer, 2021-11-24:
That very much sounds like a bug in Safari

maciejtarmas, 2021-11-24:
@Hypfer No, no. I was talking about the desktop version of Firefox. I won’t be home until later this week so I can’t check if starting a new clean up fixes this. That’s some Safari bug obviously, but I’ll take a look at memory consumption and whatnot.

fox34, 2021-11-25:
> @Hypfer It's working now. Did not change anything on the phone but I did vaccum today. Maybe due to that and the fact the map has new data it's working now? So this might be a only an issue after the upgrade when no run was made?

Can confirm that the crash is somewhat map-state-dependent. Safari on iOS 15.1 crashed yesterday while the robot was in idle mode. After starting the robot, Safari did not crash any more.

michalsliwa, 2021-11-25:
Same problem here, after checking with Safari Web Inspector i see error:
Unrecognized Content-Security-Policy directive 'worker-src'.

% curl -sI http://x.x.x.x | grep Content-Security-Policy | tr ' ' '\n'
Content-Security-Policy:
worker-src
'self';

Not sure if it's related, but apparently Safari does not support this directive

Hypfer, 2021-11-25:
https://gitlab.com/gitlab-org/gitlab-foss/-/issues/54219

The `Unrecognized Content-Security-Policy directive 'worker-src'` is normal and should not interfere with anything

maciejtarmas, 2021-11-26:
The problem seems to be related to rendering the robot's path on the map by Safari.

I sent the robot that was crashing Safari for a small spot clean and the main UI started loading on iOS, but resizing the map crashed the page again.

Note that loading the map editor @ http://robot_address/#/robot/edit_map does not cause the UI to crash and zooming and moving the map around is smooth.

I tried the current version of Safari Technology Preview on a Mac (reports as Safari 15.4, current version is 15.1) and it seems to work better, not causing huge CPU spikes, but it's still not smooth when there is the robot's path overlaid on the map. Additionally, the path is extremely blurred to a point where it's completely useless, much worse than current Safari.

It's a Safari thing for sure, but I wonder if something can be done with it.

@Hypfer , would it be possible to implement a path display on/off toggle for the UI? This way we could confirm if path rendering is the problem.

lukaswimmer, 2021-12-14:
Valetudo 2021.12.1 works on Safari in the newly released updates of iOS 15.2 and macOS 12.1.

maciejtarmas, 2021-12-14:
> Valetudo 2021.12.1 works on Safari in the newly released updates of iOS 15.2 and macOS 12.1.

No.

[image]

wienans, 2021-12-15:
For me it also looks like that it is only happening if the path is rendered. Maybe it is possible to deactivate path rendering?

Hypfer, 2021-12-15:
https://github.com/Hypfer/Valetudo/actions/runs/1584687882

This build renders paths as low-res pngs on apple mobile devices.
Please try and report back

Findus23, 2021-12-18:
I tried out the `feat_apple_png_path` branch and it seems to be working (as in the site doesn't crash on my iPad Air 2). It's still a bit slow, but I blame that on the quite old device.
Only issue is that the line drawn is really quite blurry, but it's still a huge improvement to before (and I assume without any disadvantage for non-safari-users.
[image]

wienans, 2021-12-18:
Actually I didn’t had time to test the branch but now it seems to work for me at least with the official release.
[image]
Between I updated to iOS 15.2 but didn’t checked if the issue was occurring directly after the update.

Hypfer, 2021-12-18:
What about `feat_consolidated_paths`?

Build: https://github.com/Hypfer/Valetudo/actions/runs/1596132463

Hypfer, 2021-12-27:
As far as I can tell, Apple "fixed" this crash in iOS 15.2

The path looks horrible likely due to a very low render resolution, but at least safari doesn't crash anymore.

closed 2021-12-27 by Hypfer

maciejtarmas, 2021-12-27:
> As far as I can tell, Apple "fixed" this crash in iOS 15.2

No. The screenshot is from iOS 15.2

https://github.com/Hypfer/Valetudo/issues/1232#issuecomment-993791897

sstadlberger, 2021-12-27:
> As far as I can tell, Apple "fixed" this crash in iOS 15.2
> 
> The path looks horrible likely due to a very low render resolution, but at least safari doesn't crash anymore.

I can confirm that the crash is still present in iOS 15.2. It’s not always fully reproducible as it depends on the actual path the vacuum took.

By the way, the paths always looked that pixelated for me in Safari, that was even before Safari started crashing.

I think that Safari will be fixed in one of the next releases, I tried the Safari Technology Preview and it works there without any problems while the regular Safari crashes. 

And thank you for making this great piece of software!

sstadlberger, 2021-12-29:
> What about `feat_consolidated_paths`?
> 
> Build: https://github.com/Hypfer/Valetudo/actions/runs/1596132463

Sorry for the late response. I tried it out today so it's still a bit early to be sure if this solves the issue, but so far it looks promising. Thanks again for the support!

danielk117, 2021-12-30:
> I can confirm that the crash is still present in iOS 15.2. It’s not always fully reproducible as it depends on the actual path the vacuum took.

iOS 15.2 and when i try to zoom, while it is cleaning, the webapp crashes. But only on my Dreame Z10. 
On my Roborock S5 this doesn't happen.

> By the way, the paths always looked that pixelated for me in Safari, that was even before Safari started crashing.

Same for me. 

> What about feat_consolidated_paths?
>
> Build: https://github.com/Hypfer/Valetudo/actions/runs/1596132463

I will give a try later.

lukaswimmer, 2021-12-30:
The shortly posted build works always fine on my D9. iOS 15.2 is used. 

I don’t know if it helps, but I realised the following: In release 2021.11.0 the renderer was changed. The path in the new UI of 2021.10.0 did not look pixelated, while it was pixelated in the old UI. That behaviour changed with 2021.11.0 where all UI paths were pixelated. 

2021.10.0 on iOS 15.2:
[image]

2021.12.1 on iOS 15.2:
[image]

danielk117, 2021-12-30:
> What about `feat_consolidated_paths`?
> 
> Build: https://github.com/Hypfer/Valetudo/actions/runs/1596132463

I replaced the binary. Seems to fix the crashes, but the path is still looks horrible... 😞 

[image]

maciejtarmas, 2021-12-31:
2022.01.0 doesn't crash Safari on iOS anymore and doesn't cause CPU spikes on macOS on current Safari Version 15.2 (17612.3.6.1.6) and Safari Technology Preview Release 137 (Safari 15.4, WebKit 17613.1.11.8). The path still looks like crap no matter what version of Safari or platform is used, but I can live with it. Thanks 👍

Again, could someone running GNOME Linux Desktop check how the UI is rendered on GNOME Web? Both Safari and GNOME Web use the WebKit engine, so we could at least check if the bug is specific to Safari or is there something wrong with WebKit, so we can file a bug report in an appropriate place.

Screenshot taken on an iPad Something running iOS 15.
[image]
2

Findus23, 2021-12-31:
> Again, could someone running GNOME Linux Desktop check how the UI is rendered by GNOME Web?

Both GNOME web (41.3) and Konqueror (21.08.2) are displaying the page (on 2022.01.0) perfectly fine and look identical to Firefox.

lukaswimmer, 2022-03-26:
In the recently released Safari Techology Preview 142, the blurred lines are normal lines again. So it seems that the bug could be solved by Apple in one of the next official releases.

lukaswimmer, 2022-06-07:
Path looks normal again in iOS 16 beta.