Hypfer/Valetudo #1233: home button not clickable

badsmoke, 2021-11-24:
## Describe the bug

home button not clickable in roborock gen1 version, in old frontent it still works
## To Reproduce

### Vacuum Model

roborock? mi robot v1
### Valetudo Version

 2021.1.1

## Additional context

for robots without persistent maps it would be nice to be able to save the current map and have it restored automatically on each homing

Hypfer, 2021-11-24:
Executing the `home` command is only possible while `idle` or in  `error`.

You have to press pause or stop while the robot is moving to enable the `home` button.

closed 2021-11-24 by Hypfer

badsmoke, 2021-11-24:
if you press pause or stop before it does not work either

Hypfer, 2021-11-24:
It might take a moment for the state to be updated

thinkl33t, 2021-11-24:
I have tested this on my V1 in both 2021.10.0 and 2021.11.1 and can't replicate it.  

The Home button is greyed out while the robot is moving, but becomes usable when the robot is paused or stopped.

Please can you send a screenshot / screen recording of the buttons?

badsmoke, 2021-11-24:
with my v1 robot it worked last time now

with one of my v2 robots the error is there right now.

[image]

TTR6214, 2021-12-19:
I have the same problem with rockrobo.vacuum.v1.
In the web UI, I press "Start", then "Pause", I cannot press "Home"
I press "Start", then "Stop", I can press "Home"
I press "Start", then "Pause", then "Stop", I cannot press "Home"

However, using HA and MQTT:
I use "vacuum.start", then "vacuum.pause". I can use "vacuum.return_to_base"
I use "vacuum.start", then "vacuum.stop". I can use "vacuum.return_to_base"

_Subsidiary question:_ why can't we use the basic control "Home" when the robot is cleaning?