Hypfer/Valetudo #1234: Updating via the built in updater failed

LouiS22, 2021-11-24:
## Describe the bug

-->

### Screenshots

### Vacuum Model
Roborock S5

### Valetudo Version
2021.11.0

## Expected behavior

## Additional context

the log says the following: 

[2021-11-24T14:52:05.470Z] [ERROR] Unexpected error during apply Error: ENOSPC: no space left on device, copyfile '/dev/shm/b6dd5d80-b5b1-40ee-b0df-bdbb39f74935' -> '/usr/local/bin/valetudo'
    at Object.copyFileSync (node:fs:2800:3)
    at Updater.apply (/snapshot/Valetudo/backend/lib/updater/Updater.js:112:12)
    at Updater.triggerApply (/snapshot/Valetudo/backend/lib/updater/Updater.js:100:14)
    at /snapshot/Valetudo/backend/lib/webserver/UpdaterRouter.js:41:42
    at Layer.handle [as handle_request] (/snapshot/Valetudo/node_modules/express/lib/router/layer.js:95:5)
    at next (/snapshot/Valetudo/node_modules/express/lib/router/route.js:137:13)
    at Route.dispatch (/snapshot/Valetudo/node_modules/express/lib/router/route.js:112:3)
    at Layer.handle [as handle_request] (/snapshot/Valetudo/node_modules/express/lib/router/layer.js:95:5)
    at /snapshot/Valetudo/node_modules/express/lib/router/index.js:281:22
    at Function.process_params (/snapshot/Valetudo/node_modules/express/lib/router/index.js:335:12)
    at next (/snapshot/Valetudo/node_modules/express/lib/router/index.js:275:10)
    at Function.handle (/snapshot/Valetudo/node_modules/express/lib/router/index.js:174:3)
    at router (/snapshot/Valetudo/node_modules/express/lib/router/index.js:47:12)
    at Layer.handle [as handle_request] (/snapshot/Valetudo/node_modules/express/lib/router/layer.js:95:5)
    at trim_prefix (/snapshot/Valetudo/node_modules/express/lib/router/index.js:317:13)
    at /snapshot/Valetudo/node_modules/express/lib/router/index.js:284:7 {
  errno: -28,
  syscall: 'copyfile',
  code: 'ENOSPC',
  path: '/dev/shm/b6dd5d80-b5b1-40ee-b0df-bdbb39f74935',
  dest: '/usr/local/bin/valetudo'
}
[2021-11-24T14:58:09.664Z] [ERROR] Unexpected error during apply Error: ENOENT: no such file or directory, unlink '/usr/local/bin/valetudo'
    at Object.unlinkSync (node:fs:1718:3)
    at Updater.apply (/snapshot/Valetudo/backend/lib/updater/Updater.js:111:12)
    at Updater.triggerApply (/snapshot/Valetudo/backend/lib/updater/Updater.js:100:14)
    at /snapshot/Valetudo/backend/lib/webserver/UpdaterRouter.js:41:42
    at Layer.handle [as handle_request] (/snapshot/Valetudo/node_modules/express/lib/router/layer.js:95:5)
    at next (/snapshot/Valetudo/node_modules/express/lib/router/route.js:137:13)
    at Route.dispatch (/snapshot/Valetudo/node_modules/express/lib/router/route.js:112:3)
    at Layer.handle [as handle_request] (/snapshot/Valetudo/node_modules/express/lib/router/layer.js:95:5)
    at /snapshot/Valetudo/node_modules/express/lib/router/index.js:281:22
    at Function.process_params (/snapshot/Valetudo/node_modules/express/lib/router/index.js:335:12)
    at next (/snapshot/Valetudo/node_modules/express/lib/router/index.js:275:10)
    at Function.handle (/snapshot/Valetudo/node_modules/express/lib/router/index.js:174:3)
    at router (/snapshot/Valetudo/node_modules/express/lib/router/index.js:47:12)
    at Layer.handle [as handle_request] (/snapshot/Valetudo/node_modules/express/lib/router/layer.js:95:5)
    at trim_prefix (/snapshot/Valetudo/node_modules/express/lib/router/index.js:317:13)
    at /snapshot/Valetudo/node_modules/express/lib/router/index.js:284:7 {
  errno: -2,
  syscall: 'unlink',
  code: 'ENOENT',
  path: '/usr/local/bin/valetudo'
}

Hypfer, 2021-11-24:
`/usr/local/bin/` is full?

`df` output?

LouiS22, 2021-11-24:
/usr/local/bin$ df
Filesystem           1K-blocks      Used Available Use% Mounted on
rootfs                  173533    154769      9804  94% /
/dev/root               173533    154769      9804  94% /
devtmpfs                    64         0        64   0% /dev
none                    255268        20    255248   0% /run
tmpfs                    30720       300     30420   1% /var
tmpfs                       64         0        64   0% /dev
tmpfs                    30720      1580     29140   5% /tmp
tmpfs                   102400      2560     99840   3% /run/shm
/dev/mmcblk0p1         1536048    205072   1252948  14% /mnt/data
/dev/mmcblk0p6           63461      5412     54773   9% /mnt/default
/dev/mmcblk0p10         173533    137813     26760  84% /mnt/updbuf
/dev/mmcblk0p11          15863      2193     12851  15% /mnt/reserve

Hypfer, 2021-11-24:
`rootfs 173533 154769 9804 94% /`

Why is your rootfs this full?

Here it looks like this:
```
[root@rockrobo ~]# df
Filesystem           1K-blocks      Used Available Use% Mounted on
rootfs                  505764    138475    341177  29% /
/dev/root               505764    138475    341177  29% /
devtmpfs                    64         0        64   0% /dev
none                    255268        20    255248   0% /run
tmpfs                    30720       388     30332   1% /var
tmpfs                       64         0        64   0% /dev
tmpfs                    30720         4     30716   0% /tmp
tmpfs                   102400      2564     99836   3% /run/shm
/dev/mmcblk0p1         1584408    448640   1055284  30% /mnt/data
/dev/mmcblk0p6           63461     13408     46777  22% /mnt/default
/dev/mmcblk0p10         505764    139257    340395  29% /mnt/updbuf
/dev/mmcblk0p11          15863      1943     13101  13% /mnt/reserve
```

The updater should be able to handle that gracefully though

LouiS22, 2021-11-24:
I don't know :)

Hypfer, 2021-11-24:
You've probably built your firmware image without checking the `Resize Rootfs to 510MByte checkbox`.

Please [upgrade as it was done previously](https://valetudo.cloud/pages/general/upgrading.html) and don't forget to check that box.

Starting with c298e983, the updater will no longer allow updating in such situations

closed 2021-11-24 by Hypfer

LouiS22, 2021-11-24:
Ok, I think I found something. /mnt/data contains a hidden folder .temp with a 80 mb .pkg file inside. Is that ok?

Hypfer, 2021-11-24:
Huh. Okay. That's the first time that I'm seeing that file

```
[root@rockrobo .temp]# ls -lah
drwxr-xr-x    2 root     root        4.0K Feb 20  2021 .
drwxr-xr-x    8 root     root        4.0K Aug 31 16:36 ..
-rw-r--r--    1 root     root       74.5M Feb 20  2021 Update.pkg
```

LouiS22, 2021-11-24:
OK. Will compile a new firmware and see what happens.

LouiS22, 2021-11-24:
Problem resolved, Resize Rootfs to 510MByte was the culprit.

einfacharthur, 2021-12-14:
I'd like to see this in FAQ. I had the same problem with OTA. 
Fixed this by building new image with checkbox `Resize Rootfs to 510MByte` and flashing via SSH.

https://builder.dontvacuum.me/_s5.html
https://valetudo.cloud/pages/general/upgrading.html

`[root@rockrobo ~]# df
Filesystem           1K-blocks      Used Available Use% Mounted on
rootfs                  505764    138291    341361  29% /
/dev/root               505764    138291    341361  29% /
devtmpfs                    64         0        64   0% /dev
none                    255268        20    255248   0% /run
tmpfs                    30720       168     30552   1% /var
tmpfs                       64         0        64   0% /dev
tmpfs                    30720      1484     29236   5% /tmp
tmpfs                   102400      2568     99832   3% /run/shm
/dev/mmcblk0p1         1584408    296416   1207508  20% /mnt/data
/dev/mmcblk0p6           63461      5420     54765   9% /mnt/default
/dev/mmcblk0p10         173533    138455     26118  84% /mnt/updbuf
/dev/mmcblk0p11          15863      1304     13740   9% /mnt/reserve
`

that helped :)

liquidbear99, 2021-12-16:
> 

Does this fix mean that the option will be applied by default in the subsequent releases?

rimrul, 2021-12-16:
> Does this fix mean that the option will be applied by default in the subsequent releases and upgrades using the GUI? Or will I need to continue using the old way (after fixing this by building the new image with the box checked to resize)?

You missunderstood what the updater does. It only updates valetudo without replacing the underlying firmware. So it doesn't touch what that checkbox does.

No re-applying (manual or automatic) needed.

liquidbear99, 2021-12-16:
> You missunderstood what the updater does. It only updates valetudo without replacing the underlying firmware. So it doesn't touch what that checkbox does.
>
> No re-applying (manual or automatic) needed.

Ah I see. So I just need to fix that once and it shouldn't be a problem in the future when I update valetudo since I won't need up replace the firmware again.

Hypfer, 2021-12-16:
>  (and I won't need up replace the firmware again since my firmware now has that checked).
Possibly, but there are no warranties whatsoever for that.