Hypfer/Valetudo #1240: Valetudo Crashing on Dreame D9

luke-777, 2021-11-26:
I have the same problem as in issue: https://github.com/Hypfer/Valetudo/issues/1190
The robot crashes after 1-2 minutes, without having a big map.

### Vacuum Model
Dreame D9

### Valetudo Version
2021.11.1

```/tmp/log/ava_status.log```:
```
Name:   ava
Umask:  0022
State:  S (sleeping)
Tgid:   1228
Ngid:   0
Pid:    1228
PPid:   1
TracerPid:      0
Uid:    0       0       0       0
Gid:    0       0       0       0
FDSize: 32
Groups:
VmPeak:   164140 kB
VmSize:   163304 kB
VmLck:         0 kB
VmPin:         0 kB
VmHWM:     54104 kB
VmRSS:     54104 kB
RssAnon:           31352 kB
RssFile:           22752 kB
RssShmem:              0 kB
VmData:   125216 kB
VmStk:       308 kB
VmExe:       132 kB
VmLib:     31256 kB
VmPTE:       100 kB
VmPMD:         0 kB
VmSwap:        0 kB
Threads:        11

```

Hypfer, 2021-11-26:
Please elaborate what "crashes" means exactly

luke-777, 2021-11-26:
Connection is lost, no movement, I have to restart the robot manually

Hypfer, 2021-11-26:
What is the manufacturing date of your D9 (check the packaging) and which FW version are you using?

I'm fairly certain that this is not Valetudo going OOM

luke-777, 2021-11-26:
01/2021, Firmware Version: 1100

Hypfer, 2021-11-26:
Mind sharing Valetudo logs?

luke-777, 2021-11-26:
Same problem as in the other issue, no logs after the crash.
But I found following debug-message:
```[DEBUG] GC forced at 52170752 bytes RSS was unsuccessful.```
Could this be a memory problem?

Hypfer, 2021-11-26:
As this conversation is a first-level ping pong game, I'm fairly certain that it's a user error.

You're also not follwing instructions correctly. Also, this is not a confirmed bug so this is the wrong place to discuss it.

closed 2021-11-26 by Hypfer

luke-777, 2021-11-26:
Sorry, here's the log, hope it can help.
It's freshly rooted, I never used the original App.

```[1970-01-01T00:00:25.162Z] [INFO] Set Logfile to /tmp/valetudo.log
[1970-01-01T00:00:25.179Z] [INFO] Autodetected DreameD9ValetudoRobot
[1970-01-01T00:00:25.293Z] [INFO] Starting Valetudo 2021.11.1
[1970-01-01T00:00:25.295Z] [INFO] Commit ID: 5cf5e5475476e3b930421f14eb95f343f52b292a
[1970-01-01T00:00:25.296Z] [INFO] Configuration file: /data/valetudo_config.json
[1970-01-01T00:00:25.297Z] [INFO] Logfile: /tmp/valetudo.log
[1970-01-01T00:00:25.298Z] [INFO] Robot: Dreame D9 (DreameD9ValetudoRobot)
[1970-01-01T00:00:25.298Z] [INFO] JS Runtime Version: v16.10.0
[1970-01-01T00:00:25.299Z] [INFO] Arch: arm
[1970-01-01T00:00:25.300Z] [INFO] Max Heap Size: 38 MiB
[1970-01-01T00:00:25.301Z] [INFO] Node Flags: --expose-gc --max-heap-size=38
[1970-01-01T00:00:25.312Z] [INFO] Autogenerated System ID: ExpertIntentionalFinch
[1970-01-01T00:00:25.325Z] [INFO] DeviceId 
[1970-01-01T00:00:25.326Z] [INFO] IP 127.0.0.1
[1970-01-01T00:00:25.326Z] [INFO] CloudSecret 
[1970-01-01T00:00:25.327Z] [INFO] LocalSecret 
[1970-01-01T00:00:25.329Z] [INFO] Firmware Version: 1100
[1970-01-01T00:00:25.330Z] [INFO] MCU Version: 0.9.1
[1970-01-01T00:00:25.333Z] [INFO] Current rootfs: rootfs2 (nand0p6)
[1970-01-01T00:00:38.237Z] [INFO] Valetudo can be reached via: valetudo-expertintentionalfinch.local
[1970-01-01T00:00:38.275Z] [INFO] Setting /proc/self/oom_score_adj to 666. Previous value: 0
[1970-01-01T00:00:38.292Z] [INFO] Dummycloud is spoofing 127.0.0.1:8053 on 127.0.0.1:8053
[1970-01-01T00:00:38.304Z] [INFO] Webserver running on port 80
[1970-01-01T00:00:38.317Z] [INFO] SSDP/UPnP advertisement started
[1970-01-01T00:00:38.333Z] [INFO] Map Upload Server running on port 8079
[1970-01-01T00:00:38.357Z] [INFO] Bonjour service "Valetudo D9 Web" with type "http" started
[1970-01-01T00:00:38.358Z] [INFO] Bonjour service "Valetudo D9" with type "valetudo" started
[1970-01-01T00:00:38.367Z] [WARN] Error during time sync: Error: Timeout waiting for NTP response from pool.ntp.org:123
    at Object.getNetworkTime (/snapshot/Valetudo/node_modules/@destinationstransfers/ntp/index.js:64:11)
    at async NTPClient.pollTime (/snapshot/Valetudo/backend/lib/NTPClient.js:56:36)
[1970-01-01T00:00:50.207Z] [INFO] Cloud connected
[2021-11-26T11:12:02.003Z] [INFO] Successfully set the robot time via NTP to 2021-11-26T11:12:02.895Z
```

Hypfer, 2021-11-26:
> [1970-01-01T00:00:25.329Z] [INFO] Firmware Version: 1100

is good. That means that the update was actually installed which sometimes needs additional wiggling on the D9.
There's also no idication of valetudo killing itself so it's not leaking any memory either.

Are you sure that Valetudo isn't running anymore? `ps | grep valetudo`?

Please try updating the mcu firmware by downloading the dustbuilder firmware but running `install-mcufw.sh` instead.

luke-777, 2021-11-26:
I've downloaded and reinstalled mcufw, it's still "MCU Version: 0.9.1".
5mins after crash, the robot seems to restart. Valetudo is then up again and reachable.
I have to connect the robot via serial connection after Crash and open logs to see if Valetudo still runs, right?

Hypfer, 2021-11-26:
> it's still "MCU Version: 0.9.1".

Yep. We curently don't have any way of asking the mcu which fw version it is running unfortunately. That string is taken from the /etc/os-release json.

> 5mins after crash, the robot seems to restart.

Sounds like a hardware watchdog kicking in

luke-777, 2021-11-26:
Solution: I reinstalled (install.sh) the complete package in home and tmp again. Now I works perfect! Maybe something went wrong the first time.
Thank you for your quick support!

tdehaeze, 2022-01-15:
I @luke-777 , it seems I have the same problem.
What do you mean by reinstalling in **home and tmp** again?
Thanks

luke-777, 2022-01-15:
I repeated the two installaion steps from the instruction: https://valetudo.cloud/pages/general/rooting-instructions.html:

> After the successful download, make sure that the robot is docked, untar (tar -xvzf dreame.vacuum.pxxxx_fw.tar.gz) it and execute the ./install.sh script. The robot will then reboot on its own and greet you with a shell mentioning the Dustbuilder in the MOTD on successful root.

> Switch to the tmp folder cd /tmp and repeat the previous steps (from wget to install.sh) to also install the firmware on the second partition which you are now booted to.

mcpzi, 2022-01-25:
Hi @luke-777 and @tdehaeze, I'm having the same problem with a Z10 Pro... I can't seem to find any useful log on what causes the crash.

I've tried to reinstall the firmware from dustbuilder but it does not fix the issue.

Here's the sequence:
- Launch a segment cleaning or mapping pass
- Robot starts and stops less than a minute after
- Robot does not respond to ping/ssh/webserver
- Robot reboots a few minutes later