Hypfer/Valetudo #1241: Name of segments not displayed on map

JayFoxRox, 2021-11-27:
## Describe the bug

Since upgrading to 2021.11.1, the segment names are no longer displayed on the map. Only a triangle is displayed.
This is probably caused by merging the 3 separate views from earlier versions, some of which also didn't display named segments.

Named segments seem to have little purpose now, even though they were a good feature to make it easier to read the map.
Names should be displayed again.

## To Reproduce

Steps to reproduce the behavior:

1. Go to the "Home" screen
2. No names are displayed for rooms

Tested using Firefox 94.0.1 (64-bit) on Linux.

### Screenshots

[image]

### Vacuum Model

Dreame D9 (probably generic problem).

## Expected behavior

Names for each named segment should be displayed.

Hypfer, 2021-11-27:
This behaves like it did in the old UI, meaning that the names appear after a certain zoom level to not clutter the map too much.

https://github.com/Hypfer/Valetudo/blob/660cafd9e4ff1a400f19d90391e2c640502c7f8d/frontend/src/map/structures/map_structures/SegmentLabelMapStructure.ts#L67-L74

closed 2021-11-27 by Hypfer

JayFoxRox, 2021-11-27:
> the names appear after a certain zoom level

I didn't know there was a zoom. I suspected there was some zoom, because the old frontend was barely usable on my machine due to a tiny map.

For context: I'm on a Thinkpad laptop with disabled touchpad. I don't have a mousewheel and no multi-touch.
I'd need zoom buttons or some hotkey (which should be displayed somewhere in the UI).

For other maps, I usually use keyboard keys "+"/"-" to zoom or buttons in the UI; arrow keys or mouse-drag to move. Some maps also allow double-click to zoom, or shift-box-drag to zoom to a region - however, nothing like this seems to work in Valetudo.

Hypfer, 2021-11-27:
Why don't you have a mousewheel?

I too am using thinkpads with disabled touchpads. On a linux-based operating system, it is possible to have the middle trackpoint button act like both a middle-mouse-button as well as scroll-enable-button simultaneously.
Windows doesn't allow that unfortunately

JayFoxRox, 2021-11-27:
> Why don't you have a mousewheel?

Because I don't need it.
I'm using this laptop for 3D work in Blender and similar software, and also work on map-rendering for a living (I'm a graphics programmer). None of the software (3D modelling, GIS, custom) I use has issues with my current setup.
\- Valetudo is the only app which doesn't seem to have an alternate control scheme.

The touchpad is disabled because I'd accidentally touch it with my palm + I don't like the feel of it. I use the trackpoint exclusively. Even with the touchpad, I'd probably disable the scrollwheel emulation because this model does not have a marked region on the trackpad and I don't like the ambiguity where I accidentally move my finger into the scrollwheel region.

> On a linux-based operating system with Xorg, it is possible to have the middle trackpoint button act like both a middle-mouse-button as well as scroll-enable-button simultaneously.

I don't use middle-mousebutton-hold as scrollwheel-toggle because some software I use needs middle-mousebutton-drag.
In web-browsers I scroll using auto-scroll (Firefox `about:config`: `general.autoScroll`).
So I middle click which turns the mouse-cursor into an anchor, and moving the mouse up/down causes continuous scroll, which is ideal for reading long articles.

Hypfer, 2021-11-27:
https://xkcd.com/1172/

JayFoxRox, 2021-11-27:
> https://xkcd.com/1172/

I get where you are coming from, but I don't think this is such a niche setup after all?
I don't think this is such an outlandish request.

---

I also got curious about my other machine, a Macbook Pro 2017: And I also have problems with the map navigation and map-labels on my macOS machine.

Again, it doesn't have a scrollwheel, but I can zoom using multi-touch gestures on the touchpad.
However, the map is *very* sensitive and moving my fingers barely at all will cause a very high amount of zooming.

Here's a video with 2 of my fingers, *stationary* on the trackpad, only varying the pressure (= not even moving my fingers at all, but just rotating them slightly):
I hope you can tell *how* sensitive the scrolling is.

https://user-images.githubusercontent.com/360330/143675219-a78882d4-b884-41e6-a384-cfd93032858e.mov

The zoom level that is required to get the room labels makes it impossible to see the room layout:

[image]
[image]
[image]

Getting the zoom to this level is very tricky on the macbook touchpad, and the map is almost unusable at this zoom level (labels only show up when it's too close to have spatial context).

Hypfer, 2021-11-27:
ca946009 should fix your zoom issue.

As it turns out, `WheelEvent.deltaY` is not a constant value on all devices.