Hypfer/Valetudo #1243: Redudant UI to check for Valetudo version updates

JayFoxRox, 2021-11-27:
## Describe the bug

With the introduction of the updater, the "About" screen "Latest release" box information should be moved / migrated into the updater.

I already had a version which had the updater (I think), but I still updated manually because I found the "Latest release" box in the updater, which did not allow me to update directly, so I figured there was no auto-updater.
Only after manually upgrading valetudo using SSH, I noticed the "Updater" tab which I probably could have used before.

Even in this new version, the version check / updating is split across 2 UI tabs.
This is redundant and confusing

### Screenshots

[image]
[image]

### Vacuum Model

Dreame D9 (probably a generic problem)

### Valetudo Version

2021.11.1

## Expected behavior

The additional information from "Latest release" information should migrate into the updater; so it also displays the dates of the installed / new version + the changelog link.
The about dialog version should hyper-link to the new "Updater".

Hypfer, 2021-11-27:
This is not a bug.
This is something you think that would make sense.

You're really pushing it man.

closed 2021-11-27 by Hypfer

Hypfer, 2021-11-27:
253d9f99