Hypfer/Valetudo #1246: Time related values in "Consumables" are not explained

JayFoxRox, 2021-11-27:
## Describe the bug

The values have no explanation; I'm not sure what those values mean.

- It's not clear wether they'll go up (amount of time used) or down (remaining time).
- It's not clear if good or low values are bad.
- It's not clear what will happen when resetting these values (will they become 0? will they be a device specific value? ...)

## To Reproduce

1. Go to "Consumables"

### Screenshots

[image]

### Vacuum Model

Dreame D9 (probably a generic problem)

### Valetudo Version

2021.11.1

## Expected behavior

There should be a description text and / or icons to inform the user what these values mean and when they should replace their consumables.
Different "Health" bars could indicate how urgent a change is.

The date of the last reset could be displayed so the user is aware when the tracking of each consumable started.

Hypfer, 2021-11-27:
If you don't stop opening bogus bug reports, you will be banned.

closed 2021-11-27 by Hypfer

JayFoxRox, 2021-11-27:
How is this not a bug regarding user experience? Where or how else should I report such problems?
Maybe I just don't understand how you operate this repository (despite having followed it for the last couple of weeks and having plenty of experience on GitHub).

Yes, I did open a couple of issues and discussions at the same time, but this is not because those are bogus reports, but just because I finally found time to do so / things occurred to me after the recent upgrade.

I totally share your views in https://github.com/Hypfer#about-me, but I honestly think you are mistreating a user and potential contributor here.

Hypfer, 2021-11-27:
The issues section is for proven bugs only.
e.g.

"Pressing this crashes the process"
or
"This error message appears and then the robot explodes on every tenth occasion"

for everything else there's the discussion section.
UX issues don't fit this definition of a bug here as there is no such thing as "the" correct UX. You ask ten people you get 15 different opinions.

I do appreciate the UX feedback and the time put into that but please adhere to the process established here and put them in the discussion section.

Hypfer, 2021-11-27:
After spending some time in the ~~shower~~ **chamber of understanding**, I've had an idea how to approach this.

What about having one of those questionmark in a circle icons somewhere on the right which opens a modal explaining consumables and what to do with them?

That should work on both desktop and mobile

Hypfer, 2021-11-27:
d1388f12