Hypfer/Valetudo #1249: Viomi V7 memory overflow

Omrigan, 2021-11-27:
## Describe the bug
Valetudo spontaneously shuts down with a following error message:

> Sat Nov 27 03:55:25 2021 daemon.err valetudo[702]: [2021-11-27T03:55:25.221Z] [ERROR] Valetudo is currently taking up 179138560 bytes which is more than 1/3 of available system memory. To ensure safe robot operation, it will now shutdown.
 

## To Reproduce
Just leave it as it is. In my case it happens i n 2-4 hours after I `/etc/init.d/valetudo restart`

### Vacuum Model
Viomi V7

### Valetudo Version
2021.11.0 

arm7-lowmem binary

## Expected behavior
Not crash

## Additional context

Full log here. 

Also, the fact that it is arguing about UTC - is it normal? `date` outputs UTC. Should I create another issue?

<details>
  <summary>valetudo.log</summary>

```
Sat Nov 27 03:26:37 2021 daemon.err valetudo[702]: [2021-11-27T03:26:37.480Z] [WARN] Error while adjusting timezone to UTC
Sat Nov 27 03:27:08 2021 daemon.err valetudo[702]: [2021-11-27T03:27:08.479Z] [WARN] Error while adjusting timezone to UTC
Sat Nov 27 03:27:38 2021 daemon.err valetudo[702]: [2021-11-27T03:27:38.480Z] [WARN] Error while adjusting timezone to UTC
Sat Nov 27 03:28:09 2021 daemon.err valetudo[702]: [2021-11-27T03:28:09.479Z] [WARN] Error while adjusting timezone to UTC
Sat Nov 27 03:28:39 2021 daemon.err valetudo[702]: [2021-11-27T03:28:39.479Z] [WARN] Error while adjusting timezone to UTC
Sat Nov 27 03:29:09 2021 daemon.err valetudo[702]: [2021-11-27T03:29:09.480Z] [WARN] Error while adjusting timezone to UTC
Sat Nov 27 03:29:39 2021 daemon.err valetudo[702]: [2021-11-27T03:29:39.486Z] [WARN] Error while adjusting timezone to UTC
Sat Nov 27 03:30:10 2021 daemon.err valetudo[702]: [2021-11-27T03:30:10.479Z] [WARN] Error while adjusting timezone to UTC
Sat Nov 27 03:30:40 2021 daemon.err valetudo[702]: [2021-11-27T03:30:40.804Z] [WARN] Error while adjusting timezone to UTC
Sat Nov 27 03:31:11 2021 daemon.err valetudo[702]: [2021-11-27T03:31:11.479Z] [WARN] Error while adjusting timezone to UTC
Sat Nov 27 03:31:41 2021 daemon.err valetudo[702]: [2021-11-27T03:31:41.671Z] [WARN] Error while adjusting timezone to UTC
Sat Nov 27 03:32:12 2021 daemon.err valetudo[702]: [2021-11-27T03:32:12.486Z] [WARN] Error while adjusting timezone to UTC
Sat Nov 27 03:32:42 2021 daemon.err valetudo[702]: [2021-11-27T03:32:42.492Z] [WARN] Error while adjusting timezone to UTC
Sat Nov 27 03:33:13 2021 daemon.err valetudo[702]: [2021-11-27T03:33:13.486Z] [WARN] Error while adjusting timezone to UTC
Sat Nov 27 03:33:43 2021 daemon.err valetudo[702]: [2021-11-27T03:33:43.479Z] [WARN] Error while adjusting timezone to UTC
Sat Nov 27 03:34:13 2021 daemon.err valetudo[702]: [2021-11-27T03:34:13.486Z] [WARN] Error while adjusting timezone to UTC
Sat Nov 27 03:34:43 2021 daemon.err valetudo[702]: [2021-11-27T03:34:43.485Z] [WARN] Error while adjusting timezone to UTC
Sat Nov 27 03:35:14 2021 daemon.err valetudo[702]: [2021-11-27T03:35:14.479Z] [WARN] Error while adjusting timezone to UTC
Sat Nov 27 03:35:44 2021 daemon.err valetudo[702]: [2021-11-27T03:35:44.479Z] [WARN] Error while adjusting timezone to UTC
Sat Nov 27 03:36:15 2021 daemon.err valetudo[702]: [2021-11-27T03:36:15.480Z] [WARN] Error while adjusting timezone to UTC
Sat Nov 27 03:36:45 2021 daemon.err valetudo[702]: [2021-11-27T03:36:45.480Z] [WARN] Error while adjusting timezone to UTC
Sat Nov 27 03:37:16 2021 daemon.err valetudo[702]: [2021-11-27T03:37:16.479Z] [WARN] Error while adjusting timezone to UTC
Sat Nov 27 03:37:46 2021 daemon.err valetudo[702]: [2021-11-27T03:37:46.516Z] [WARN] Error while adjusting timezone to UTC
Sat Nov 27 03:38:17 2021 daemon.err valetudo[702]: [2021-11-27T03:38:17.479Z] [WARN] Error while adjusting timezone to UTC
Sat Nov 27 03:38:47 2021 daemon.err valetudo[702]: [2021-11-27T03:38:47.479Z] [WARN] Error while adjusting timezone to UTC
Sat Nov 27 03:39:17 2021 daemon.err valetudo[702]: [2021-11-27T03:39:17.494Z] [WARN] Error while adjusting timezone to UTC
Sat Nov 27 03:39:47 2021 daemon.err valetudo[702]: [2021-11-27T03:39:47.480Z] [WARN] Error while adjusting timezone to UTC
Sat Nov 27 03:40:18 2021 daemon.err valetudo[702]: [2021-11-27T03:40:18.480Z] [WARN] Error while adjusting timezone to UTC
Sat Nov 27 03:40:48 2021 daemon.err valetudo[702]: [2021-11-27T03:40:48.480Z] [WARN] Error while adjusting timezone to UTC
Sat Nov 27 03:41:19 2021 daemon.err valetudo[702]: [2021-11-27T03:41:19.479Z] [WARN] Error while adjusting timezone to UTC
Sat Nov 27 03:41:49 2021 daemon.err valetudo[702]: [2021-11-27T03:41:49.486Z] [WARN] Error while adjusting timezone to UTC
Sat Nov 27 03:42:20 2021 daemon.err valetudo[702]: [2021-11-27T03:42:20.479Z] [WARN] Error while adjusting timezone to UTC
Sat Nov 27 03:42:50 2021 daemon.err valetudo[702]: [2021-11-27T03:42:50.677Z] [WARN] Error while adjusting timezone to UTC
Sat Nov 27 03:43:20 2021 daemon.err valetudo[702]: [2021-11-27T03:43:20.499Z] [WARN] Error while adjusting timezone to UTC
Sat Nov 27 03:43:50 2021 daemon.err valetudo[702]: [2021-11-27T03:43:50.486Z] [WARN] Error while adjusting timezone to UTC
Sat Nov 27 03:44:21 2021 daemon.err valetudo[702]: [2021-11-27T03:44:21.479Z] [WARN] Error while adjusting timezone to UTC
Sat Nov 27 03:44:51 2021 daemon.err valetudo[702]: [2021-11-27T03:44:51.486Z] [WARN] Error while adjusting timezone to UTC
Sat Nov 27 03:45:22 2021 daemon.err valetudo[702]: [2021-11-27T03:45:22.479Z] [WARN] Error while adjusting timezone to UTC
Sat Nov 27 03:45:52 2021 daemon.err valetudo[702]: [2021-11-27T03:45:52.719Z] [WARN] Error while adjusting timezone to UTC
Sat Nov 27 03:46:23 2021 daemon.err valetudo[702]: [2021-11-27T03:46:23.486Z] [WARN] Error while adjusting timezone to UTC
[... 139 lines cut ...]
Sat Nov 27 13:38:33 2021 authpriv.info dropbear[2546]: Child connection from 192.168.3.1:59570
Sat Nov 27 13:38:33 2021 authpriv.notice dropbear[2546]: Pubkey auth succeeded for 'root' with key md5 15:4c:6b:4a:47:38:1b:66:d0:e2:79:8c:2e:8c:d5:c7 from 192.168.3.1:59570
Sat Nov 27 13:38:33 2021 authpriv.info dropbear[2546]: Exit (root): Disconnect received
Sat Nov 27 13:38:41 2021 authpriv.info dropbear[2549]: Child connection from 192.168.3.1:59572
Sat Nov 27 13:38:41 2021 authpriv.notice dropbear[2549]: Pubkey auth succeeded for 'root' with key md5 15:4c:6b:4a:47:38:1b:66:d0:e2:79:8c:2e:8c:d5:c7 from 192.168.3.1:59572
```
</details>

<details>
  <summary>Top processes after restarting valetudo</summary>

```
$ ps -o pid,user,%mem,command ax   | sort -b -k3 -r  | head
 PID USER     %MEM COMMAND
 2602 root      8.6 /mnt/UDISK/valetudo
  620 root      6.0 RobotApp
  657 root      5.2 everest-server
  631 root      1.0 AuxCtrl
  683 root      0.3 /usr/sbin/upgrade -D
  675 root      0.3 log-server
  600 root      0.2 /usr/sbin/Monitor -D
  768 root      0.1 wpa_supplicant -dd -iwlan0 -Dnl80211 -c/etc/wifi/wpa_supplicant.conf
    1 root      0.1 /sbin/procd
```
</details>

Omrigan, 2021-12-03:
Temporarily disabled MQTT, seems to "fix" the issue. 

>Valetudo is currently taking up 179138560 bytes which is more than 1/3 of available system memory. To ensure safe robot operation, it will now shutdown.

Maybe we should have an option like what share of my memory I am willing to give to valetudo. Maybe if it is tweaked to 1/2, robot would still work.

Hypfer, 2022-01-25:
Did this ever happen with newer versions of valetudo?

Omrigan, 2022-02-06:
I've been using 2022.02 version for several days now, everything seems to be alright.

I'll reopen the issue if I encounter problems again.

Thanks for the fix!

closed 2022-02-06 by Omrigan