Hypfer/Valetudo #1256: New log viewer misses some lines

ccoors, 2021-12-01:
## Describe the bug
The log viewer misses a few lines:

[image]

## To Reproduce
Open the log viewer, set the log level to 'debug' and wait for a line starting with "Map upload started with". The object printed in [this line](https://github.com/Hypfer/Valetudo/blob/master/backend/lib/robots/MiioValetudoRobot.js#L83) is not displayed correctly.

Looking at the log file via SSH shows the missing lines:
[image]

### Vacuum Model
Dreame Z10 Pro, FW 1099

### Valetudo Version
2021.11.1 (Dev-build from 4c9dc3bfb99d09319749b6f26451248b2e493bb2)

## Expected behavior
The missing lines are displayed in the log.

## Additional context

Hypfer, 2021-12-01:
[image]

huh.
mind sharing the logfile?

ccoors, 2021-12-01:
Huh. Just noticed it only happens when getting new log lines via SSE. After reloading the page they are displayed correctly.

commit fd6f0e2d (2021-12-02): fix: Fix log sse endpoint

this fixes #1256
closed 2021-12-02 by Hypfer