Hypfer/Valetudo #1257: Install Process lags id_file

fabian727, 2021-12-02:
I followed the manual and got stuck at the step to install the firmware:
`mirobo --token XXXXXXXXXXXXXXXX --ip ROBOT_IP_ADDRESS update-firmware path/to/built/image.pkg`

instead of the update procedure, I got errors.
I got them fixed by creating an id file with {} as content and reference it.
Instead of above command, I would suggest to use:

`echo {} >> id_file.txt
mirobo --id-file id_file.txt --token XXXXXXXXXXXXXXXX --ip ROBOT_IP_ADDRESS update-firmware path/to/built/image.pkg`

Hypfer, 2021-12-02:
This is not the https://github.com/rytilahti/python-miio/ bugtracker

fabian727, 2021-12-02:
sry,
still great job
and fast reaction here too, nice