Hypfer/Valetudo #1270: MQTT connection fails until robot is rebooted

daniel1v, 2021-12-09:
## Describe the bug

From time to time, Valetudo is not able to reconnect to MQTT broker after losing the connection, unless the robot is manually rebooted.

## To Reproduce

Cannot be reproduced, happens randomly.

### Screenshots

###Log

`2021-12-09T03:26:49.230Z
Info
MQTT configured
2021-12-09T11:41:49.850Z
Info
Connection to MQTT broker closed
2021-12-09T11:41:50.850Z
Info
Attempting to reconnect to MQTT broker
2021-12-09T11:42:02.049Z
Info
Attempting to reconnect to MQTT broker
2021-12-09T11:42:13.495Z
Info
Connected successfully to MQTT broker
2021-12-09T11:44:14.639Z
Info
Attempting to reconnect to MQTT broker
2021-12-09T11:44:17.692Z
Info
Attempting to reconnect to MQTT broker
2021-12-09T11:44:20.764Z
Info
Attempting to reconnect to MQTT broker
2021-12-09T11:44:23.836Z
Info
Attempting to reconnect to MQTT broker
2021-12-09T11:44:26.908Z
Info
Attempting to reconnect to MQTT broker
2021-12-09T11:44:29.980Z
Info
Attempting to reconnect to MQTT broker
2021-12-09T11:44:30.082Z
Info
Connected successfully to MQTT broker
2021-12-09T11:44:30.341Z
Info
MQTT configured
2021-12-09T11:48:47.426Z
Error
MQTT error: Error: read ECONNRESET
2021-12-09T11:48:48.428Z
Info
Attempting to reconnect to MQTT broker
2021-12-09T11:50:58.745Z
Info
Attempting to reconnect to MQTT broker
2021-12-09T11:51:01.820Z
Info
Attempting to reconnect to MQTT broker
2021-12-09T11:51:04.892Z
Info
Attempting to reconnect to MQTT broker
2021-12-09T11:51:07.964Z
Info
Attempting to reconnect to MQTT broker
2021-12-09T11:51:11.036Z
Info
Attempting to reconnect to MQTT broker
2021-12-09T11:51:14.108Z
Info
Attempting to reconnect to MQTT broker
2021-12-09T11:51:17.180Z
Info
Attempting to reconnect to MQTT broker
2021-12-09T11:51:20.252Z
Info
Attempting to reconnect to MQTT broker
2021-12-09T11:51:23.324Z
Info
Attempting to reconnect to MQTT broker
2021-12-09T11:51:26.396Z
Info
Attempting to reconnect to MQTT broker
2021-12-09T11:51:29.468Z
Info
Attempting to reconnect to MQTT broker
2021-12-09T11:51:32.540Z
Info
Attempting to reconnect to MQTT broker
2021-12-09T11:51:35.612Z
Info
Attempting to reconnect to MQTT broker
2021-12-09T11:51:38.684Z
Info
Attempting to reconnect to MQTT broker
2021-12-09T11:51:41.756Z
Info
Attempting to reconnect to MQTT broker
2021-12-09T11:51:44.828Z
Info
Attempting to reconnect to MQTT broker
2021-12-09T11:51:47.900Z
Info
Attempting to reconnect to MQTT broker
2021-12-09T11:51:50.972Z
Info
Attempting to reconnect to MQTT broker
2021-12-09T11:51:54.044Z
Info
Attempting to reconnect to MQTT broker
2021-12-09T11:52:25.047Z
Info
Attempting to reconnect to MQTT broker
2021-12-09T11:52:36.412Z
Info
Attempting to reconnect to MQTT broker
2021-12-09T11:52:40.476Z
Info
Attempting to reconnect to MQTT broker
2021-12-09T11:52:43.547Z
Info
Attempting to reconnect to MQTT broker
2021-12-09T11:52:46.620Z
Info
Attempting to reconnect to MQTT broker
2021-12-09T11:53:06.108Z
Info
Attempting to reconnect to MQTT broker
2021-12-09T11:53:10.212Z
Info
Attempting to reconnect to MQTT broker
2021-12-09T11:53:13.244Z
Info
Attempting to reconnect to MQTT broker
2021-12-09T11:53:16.316Z
Info
Attempting to reconnect to MQTT broker
2021-12-09T11:53:19.388Z
Info
Attempting to reconnect to MQTT broker
2021-12-09T11:53:22.460Z
Info
Attempting to reconnect to MQTT broker
2021-12-09T11:53:24.157Z
Info
Connected successfully to MQTT broker
2021-12-09T11:53:24.266Z
Info
MQTT configured
2021-12-09T11:55:45.123Z
Error
MQTT error: Error: read ECONNRESET
2021-12-09T11:55:46.124Z
Info
Attempting to reconnect to MQTT broker
2021-12-09T11:55:46.810Z
Info
Connected successfully to MQTT broker
2021-12-09T11:55:46.896Z
Info
MQTT configured
2021-12-09T11:55:47.634Z
Info
MQTT configured
2021-12-09T12:20:58.799Z
Error
MQTT error: Error: read ECONNRESET
2021-12-09T12:20:58.812Z
Info
Connection to MQTT broker closed
2021-12-09T12:20:59.813Z
Info
Attempting to reconnect to MQTT broker
2021-12-09T12:21:01.424Z
Info
Disconnecting from the MQTT Broker...
2021-12-09T15:55:41.365Z
Info
Disconnecting from the MQTT Broker...
2021-12-09T15:55:45.240Z
Info
Disconnecting from the MQTT Broker...
2021-12-09T17:01:59.323Z
Info
Disconnecting from the MQTT Broker...
2021-12-09T17:03:17.949Z
Info
Disconnecting from the MQTT Broker...`

### Vacuum Model

Roborock S5 and Dreame Z9 Pro

### Valetudo Version

2021.12.0, but also happenend on every earlier version since approx 5 months

## Expected behavior

After losing connection to the MQTT broker, the device should properly reconnect.

## Additional context

After some days of everything working fine, suddenly the robot loses connection to  MQTT and is not able to reconnect, until I manually reboot it via power button. The network connection is fine and the issue appears when the robot is in it's dock. Valetudo web interface can still be accessed and the device is fully functional. Only MQTT does not work anymore. Disabling and re-enabling MQTT does not solve the issue, only a reboot of the robot. Rebooting my MQTT server does also not help, and other MQTT devices keep on working, so it is definitely nothing with my MQTT server.

The issue faces also on both of my robots, but randomly and not at the same time. It happens approx. 1 - 2 times a week / per robot.

According to the log, it looks like, that after some connection retries, it keeps stuck in "Disconnecting from the MQTT Broker...".

DuBass858, 2021-12-10:
The same problem viomi v7
I decide by rebooting every day according to the script

daniel1v, 2021-12-10:
> The same problem viomi v7 I decide by rebooting every day according to the script

according to which script do you mean?

DuBass858, 2021-12-10:
> > The same problem viomi v7 I decide by rebooting every day according to the script
> 
> according to which script do you mean?

by automation from iobroker (doesn't matter which one to use)
`sudo -S ssh root@vacuum reboot`

daniel1v, 2021-12-10:
> > > The same problem viomi v7 I decide by rebooting every day according to the script
> > 
> > 
> > according to which script do you mean?
> 
> by automation from iobroker (doesn't matter which one to use) `sudo -S ssh root@vacuum reboot`

Ok, I understand. Initially I thought you were referring to a script inide of Valetudo. If the issue persists, I will try your workaround. At the moment my hope is, that it gets fixed. 😊

Hypfer, 2021-12-16:
> According to the log, it looks like, that after some connection retries, it keeps stuck in "Disconnecting from the MQTT Broker...".

This logline was added in a recent release to provide diagnostics and diagnostics it did provide.

https://github.com/Hypfer/Valetudo/blob/120079dacc0b797a4707e35a41a5a5bfd83f96d6/backend/lib/mqtt/MqttController.js#L375-L377

One of these two promises isn't resolving, leading to everything being stuck forever.
Unfortunately though, it could be both

Hypfer, 2021-12-19:
ced9ee1b3312489c0c937723ba8744c21264f540 might fix this.

It is however still recommended to fix the network if `ECONNRESET` is a daily occurence.

daniel1v, 2021-12-19:
> [ced9ee1](https://github.com/Hypfer/Valetudo/commit/ced9ee1b3312489c0c937723ba8744c21264f540) might fix this.
> 
> It is however still recommended to fix the network if `ECONNRESET` is a daily occurence.

Hello @Hypfer, thank you for your answer.

Indeed the ECONNRESET appears from time to time in my logs. I don't know, if it is daily. Caould you please go a bit more into detail? What do you mean with "fix the network"?

Hypfer, 2021-12-19:
ECONNRESET means that something broke in a hard way which should not be happening in a network that is operating correctly.
It could be anything in the path between the robot and the mqtt server
- Bad wifi reception
- Bad wifi AP
- Bad wifi drivers
- Bad ethernet cables
- Bad ethernet switches
- Bad ethernet NICs
- Bad software
- etc

In any case, it's something that needs fixing

daniel1v, 2021-12-19:
Okay, I understand the point, that there could be some connection issues. 

But how can I be sure, that the issue is not the device itself or the software on the device? I have lots of network devices and services running without such issues.

Hypfer, 2021-12-19:
Because it doesn't happen for a lot of other users of this robot including the author of the software

thomaswr, 2021-12-19:
@Hypfer 
I have a similar problem with a Roborock S5 (from 2018). I'm using valetudo (2021.12.0) with Home Assistant (2021.11.05) and its Mosquitto broker addon.

It takes only hours before the connection between HA and Valetudo breaks and I can't control it from HA anymore, Valetudo works.I had this setup for a long time and it worked great but I'm using valetudo only for some months now.

I can't find anything obvious in the logs. Valetudo (or the S5) seem to restart each night at 3:00. 

How can I help to debug?

daniel1v, 2021-12-19:
> Because it doesn't happen for a lot of other users of this robot including the author of the software

Okay, then I will close this issue, as it seems to not meet these criteria. Thank you for your time.

Apocrathia, 2021-12-19:
I am also having this issue with the Dreame Z10 Pro that I just got yesterday to replace my Roborock S5 (which never had this issue). It would appear that it went dark around 0500 MST and didn't come back up until I rebooted the unit over SSH. I've set the logging level to debug (The log viewer in 2021.12.1 was a great addition), and will report back once I see it happen again.

> The following screenshot is from my Home Assistant instance where the device is connected via MQTT
> [image]

While I understand that Hypfer (Not tagging them so they don't get pinged) hasn't experienced this issue, it's clear that a subset of users are. Therefore, it will be on the community to troubleshoot this issue. So far, this issue has been reported with the following vacuums:
- Roborock S5
- Dreame Z10 Pro
- Viomi v7

luizribeiro, 2021-12-19:
I *believe* this has been happening to me when my vacuum roams between access points. I've enabled logging on both my vacuum and access points and will report back with what I learn.

daniel1v, 2021-12-19:
> I _believe_ this has been happening to me when my vacuum roams between access points. I've enabled logging on both my vacuum and access points and will report back with what I learn.

I can confirm that as a _possible_ reason. I have multiple access points which are accesible from the robots docking position. Although, the issue never appeared, while the robot was working, but it definitely has to change acces point while working. So this cannot be the only reason or does not always trigger the issue.

Apocrathia, 2021-12-19:
That's actually a great idea. I'm going to see if it happens again and then I will try restricting the vacuum to it's respective access point (I have a vacuum and AP on each floor). What access points are you guys using? That may be a common factor here.

ctvedt, 2021-12-19:
I only have a single AP, so roaming could not be the root cause. Looking forward to trying out the next release to see if the fix resolves this issue!

Apocrathia, 2021-12-21:
So, my vacuum still hasn't dropped off of MQTT since Sunday morning. I did restart my MQTT broker amidst my troubleshooting effort, and that may have been what caused the issue. Has anyone else tried restarting their broker to see if that resolves the issue?

Hypfer, 2021-12-21:
Please re-read this comment.
https://github.com/Hypfer/Valetudo/issues/1270#issuecomment-997337801

daniel1v, 2021-12-21:
Thank you, now there are no more questions.

Hypfer, 2021-12-29:
42a5613d19cd46bba0e5243f656d062f8a4d4415

This commit fixes another issue with the mqtt interface that also led to it being stuck forever.

ibanmda, 2022-01-01:
Hi!  I have this issue too. But after I read the messages of @Hypfer Hypfter, I think that my cause could be the poor wifi signal when the robot is cleaning at the other end of my house.

> 2022-01-01T10:56:26.348Z
> Error
> MQTT error: Error: read ECONNRESET
> 
> 2022-01-01T10:56:26.480Z
> Info
> Connection to MQTT broker closed
> 
> 2022-01-01T10:56:27.481Z
> Info
> Attempting to reconnect to MQTT broker
> 
> 2022-01-01T10:56:36.669Z
> Info
> Disconnecting from the MQTT Broker...

daniel1v, 2022-01-03:
2022.1.0 in fact resolves the issue, where it seems to hang at `Disconnecting from the MQTT Broker...`. 

Unfortunately, the MQTT connection still seems to fail. Now there is an infinite loop of `Attempting to reconnect to MQTT broker`. This happens still suddenly, when the robot is in the dock, where it has basically a good wifi reception and both, the robot and the broker are accessible on the network. I cannot guarantee, that there are absolutely no wifi disconnections over the whole day, but imho, the robot should be able to handle this and reconnect, even if the wifi connection drops sometimes out. Rebooting the robot solves the issue, after the reboot, the MQTT connection works instantly.

@Hypfer, how do you think about this? I reopened this issue and hope, it is ok for you. I can also assist in debugging, if someone can give me advices, where to start.

Hypfer, 2022-01-03:
Is it just logging `Attempting to reconnect to MQTT broker` without any additional connection failed messages?

daniel1v, 2022-01-03:
Yes, there was only this one line, every second or so. The log level was set to `Info`. Unfortunately, I already did a reboot, so I cannot provide further details of the log, unless the old logfile persists somwhere on the robot?

Hypfer, 2022-01-03:
Please check your broker logs and make sure that you're using mosquitto as your broker.

If it's just logging `Attempting to reconnect to MQTT broker` then it's just the mqtt library reconnect functionality doing its job with no breakage inside Valetudo. As it's constantly reconnecting without any error message such as `ECONNREFUSED` being logged, the library is likely just not received a `connack` response by the broker in time.

I suspect either a broken broker, broker misconfiguration or some other networking issues

Hypfer, 2022-01-03:
I can reproduce that behaviour by just running netcat and letting the mqtt client connect to it

`nc -l -p 1884 -k`

There seems to be a 30s timeout there though but apart from that it behaves the same

If we turn to nodejs and write a simple tcp server which instantly closes the connection, we now get exactly the described behaviour
```js
const net = require("net");
const mockServer = net.createServer();

mockServer.on("connection", con => {
    con.end();
})

mockServer.listen(1883);
```

```
[2022-01-03T10:20:42.967Z] [INFO] Attempting to reconnect to MQTT broker
[2022-01-03T10:20:43.968Z] [INFO] Attempting to reconnect to MQTT broker
[2022-01-03T10:20:44.971Z] [INFO] Attempting to reconnect to MQTT broker
[2022-01-03T10:20:45.974Z] [INFO] Attempting to reconnect to MQTT broker
[2022-01-03T10:20:46.987Z] [INFO] Attempting to reconnect to MQTT broker
[2022-01-03T10:20:47.994Z] [INFO] Attempting to reconnect to MQTT broker
[2022-01-03T10:20:49.001Z] [INFO] Attempting to reconnect to MQTT broker
[2022-01-03T10:20:50.006Z] [INFO] Attempting to reconnect to MQTT broker
```

Maybe firewall? Idk. In any case something is allowing the connection but instantly closes it without responding with a connack.
Could also be broker ACLs I guess?

closed 2022-01-03 by Hypfer

daniel1v, 2022-01-03:
Thank you for the fast reply.

1) I am using the latest eclipse mosquitto docker container.
2) The broker log does not show anything special. These lines seem to appear repeatedly: 
   ```
   1641093951: New connection from 192.168.178.37:54682 on port 1883.
   1641093952: Client valetudo_PaleLuxuriousFinch already connected, closing old connection.
   1641093952: New client connected from 192.168.178.37:54682 as valetudo_PaleLuxuriousFinch (p2, c0, k60).
   ```
  but not as often, as the reconnect message in the valetudo logs.
3) Other devices keep on working with the broker, at the same time, the issue appears on valetudo. Also, my second valetudo device.
4) As of 3)  and because the issue can only be resolved by a simple restart of the robot (and really only by that!) imho it is very likely, that the cause is on the client side. Also, a restart of the MQTT broker or the whole server, where the broker runs, does not have any effect.

So please reconsider your decision to close this issue.

Hypfer, 2022-01-03:
I don't see how Valetudo can do anything about the MQTT server closing the connection without responding with a CONNACK

schinken, 2022-01-03:
> 1. I am using the latest eclipse mosquitto docker container.

me too 

> 2. The broker log does not show anything special. These lines seem to appear repeatedly:

I only can see disconnects when I turn of my robot to clean it or when I do updates actually. Never had this reconnect-issue...

Btw, just as a side-note, i would recommend this option for your mosquitto.log, so it's more readable:
`
log_timestamp true
log_timestamp_format %Y-%m-%dT%H:%M:%S
`

Stevelriemenbill, 2022-01-03:
I also have this issue running Valetudo 2022.01.0 and the newest version of homeassistant and mosquitto as docker containers.
Same behaviour: Only a reboot of the robot makes the mqtt connection work again.
These are the log messages every 3 seconds:
```
2022-01-03T11:14:08.770Z trace
MQTT State change { prev: 'lost', new: 'lost' }
2022-01-03T11:14:09.288Z debug
GC forced at 51240960 bytes RSS was unsuccessful.
2022-01-03T11:14:09.770Z info
Attempting to reconnect to MQTT broker
```
I have no other error messages.

daniel1v, 2022-01-03:
> I don't see how Valetudo can do anything about the MQTT server closing the connection without responding with a CONNACK

Is this really true? I cannot find any hint, that the broker is closing the connection without responding. There's, for example, also the possitlity, that the server responds, but the client does not recognize it, for some reason.

Imho there is one important question which resides here: If your theory is right, that the issue is _not_ based in valetudo / the mqtt client: why is the only way to get it back working, to reboot the robot, and not the server, or anything else? from my point of view, the only logical reason for this behaviour is, that there is indeed something wrong on client side. 

I can live with the fact, that this is not considered as high priority, but I doubt, that I am the only one facing this issues and I strongly believe, that the issue can be solved on client side, because non valetudo clients do not suffer from these issue.

Please do not misunderstand me: I do not want to offense you and I love valetudo. I am only trying, to participate and make it more stable and usable for everyone (including me).

So please consider reopening the issue and rethink your theory, that the issue is server-side.

Hypfer, 2022-01-03:
> I cannot find any hint, that the broker is closing the connection without responding.

Yes. We've already established that you're refusing to read and understand my replies.
Please consider switching back to the official cloud as I'm done with dealing with your nonsense.