Hypfer/Valetudo #1273: An error occurred: No error

cradcore, 2021-12-10:
Just bought a Roborock S7 on Black Friday and am absolutely loving Valetudo so far. Thank you for your hard work. For some reason, every time I run the vacuum, I get these events that say "An error occurred: No error" Any idea why or how to fix it? Even adding a "dismiss all" button would be incredibly useful, as I have about 50 notifications from the past couple days

[image]

Hypfer, 2021-12-17:
Do you see some kind of pattern when this appears?

This is happening because the robot is in state 12 which is error
https://github.com/Hypfer/Valetudo/blob/b3063989f0210bfee901ae164e0b2fb092e3b4eb/backend/lib/robots/roborock/RoborockValetudoRobot.js#L519-L521

however the error description code is 0 which is no error
https://github.com/Hypfer/Valetudo/blob/b3063989f0210bfee901ae164e0b2fb092e3b4eb/backend/lib/robots/roborock/RoborockValetudoRobot.js#L556-L559

cradcore, 2021-12-26:
Sorry for the delay in responding, I have been traveling for the holidays. No pattern as far as I can tell, it happens seemingly randomly and a random amount of times per clean. Sometimes it happens about 5 times during a clean, somethings up to 15

Is there a way I can debug or find more info as to why it’s getting into an error state but no error description is set?

FarFarFromH0me, 2022-01-02:
Hey everybody,

just wanted to confirm this issue, also got this occasionally without any pattern.
This startet with the new UI if i remeber correctly (01.10.2021?)

Rikorose, 2022-01-07:
I get it as well, usually at the start of an action (cleaning, selected room cleaning) 2 or 3 times.

repr1sal, 2022-01-08:
Hi there!

I do confirm this issue aswell. Ich have a Roborock S7 with a "Auto-Empty Dock".

Rooted via FEL mode, shell access available
Version of Valetudo: 2022.01.0 (arm, low mem)
Firmware version: Firmware Version: 1398
JS Runtime Version: v16.10.0
Arch: arm
Max Heap Size: 34 MiB
Node Flags: --expose-gc --max-heap-size=34 --optimize-for-size --lite-mode --no-regexp-tier-up --no-expose-wasm

In the Log the Message "warn Failed to parse uploaded map" shows up without a regular basis.

In addition I Fund som info message in the log: " info
Cloud connected" and "info Unknown cloud message received: {"id":0,"result":"unknown_method"}"

`2022-01-08T09:26:57.922Z info Cloud connected
2022-01-08T09:26:58.064Z info Unknown cloud message received: {"id":0,"result":"unknown_method"}
2022-01-08T09:27:05.533Z warn Failed to parse uploaded map
2022-01-08T09:27:09.487Z warn Failed to parse uploaded map
2022-01-08T09:28:09.497Z warn Failed to parse uploaded map
2022-01-08T09:28:13.503Z warn Failed to parse uploaded map
2022-01-08T09:29:13.516Z warn Failed to parse uploaded map`

and further...

`2022-01-08T11:04:11.204Z warn Failed to parse uploaded map
2022-01-08T11:04:15.211Z warn Failed to parse uploaded map
2022-01-08T11:04:19.224Z warn Failed to parse uploaded map
2022-01-08T11:09:03.903Z info Unknown property change message received: {"did":"505936193","siid":2,"piid":4,"value":102}
2022-01-08T11:09:03.984Z info Unknown property change message received: {"did":"505936193","siid":2,"piid":4,"value":102}
2022-01-08T11:14:03.455Z info Unknown property change message received: {"did":"505936193","siid":2,"piid":4,"value":103}
2022-01-08T11:14:03.481Z info Unknown property change message received: {"did":"505936193","siid":2,"piid":4,"value":103}
2022-01-08T11:14:09.390Z info Unknown property change message received: {"did":"505936193","siid":2,"piid":4,"value":102}
2022-01-08T11:14:09.403Z info Unknown property change message received: {"did":"505936193","siid":2,"piid":4,"value":102}
2022-01-08T11:15:50.535Z info Unknown property change message received: {"did":"505936193","siid":2,"piid":4,"value":103}
2022-01-08T11:15:50.623Z info Unknown property change message received: {"did":"505936193","siid":2,"piid":4,"value":103}
2022-01-08T11:19:04.914Z info Unknown property change message received: {"did":"505936193","siid":2,"piid":4,"value":101}
2022-01-08T11:19:04.945Z info Unknown property change message received: {"did":"505936193","siid":2,"piid":4,"value":101}
2022-01-08T11:19:11.979Z info Unknown property change message received: {"did":"505936193","siid":2,"piid":4,"value":102}
2022-01-08T11:19:11.995Z info Unknown property change message received: {"did":"505936193","siid":2,"piid":4,"value":102}
2022-01-08T11:19:15.216Z info Unknown property change message received: {"did":"505936193","siid":2,"piid":4,"value":101}
2022-01-08T11:19:15.315Z info Unknown property change message received: {"did":"505936193","siid":2,"piid":4,"value":101}
2022-01-08T11:19:16.885Z info Unknown property change message received: {"did":"505936193","siid":2,"piid":4,"value":105}
2022-01-08T11:19:16.899Z info Unknown property change message received: {"did":"505936193","siid":2,"piid":4,"value":105}
2022-01-08T11:19:18.925Z info Unknown property change message received: {"did":"505936193","siid":2,"piid":4,"value":103}
2022-01-08T11:19:18.998Z info Unknown property change message received: {"did":"505936193","siid":2,"piid":4,"value":103}
2022-01-08T11:19:21.408Z info Unknown property change message received: {"did":"505936193","siid":2,"piid":4,"value":104}`

I dont't no if this is all related but that's what shows up on running the robot.

I do appreciate your help, thanks in advance.

greets

adminy, 2022-01-22:
Same here.

I noticed the log contains:
```
[2022-01-22T11:00:48.529Z] [INFO] Unknown cloud message received: {"id":0,"result":"unknown_method"}
[2022-01-22T11:03:08.886Z] [INFO] Unknown property change message received: {"did":"465596333","siid":2,"piid":4,"value":103}
```

Hypfer, 2022-01-23:
If this means nothing, we could remap it, but to be able to remap `Status Error, Error code 0` to something that makes sense, we'd need to know what we should remap that state to.

Does this happen while it is docked? Idle? Cleaning?

Also, does it actually not mean anything?

adminy, 2022-01-23:
While cleaning. never happens when docked, I can confirm that. Sometimes could happen when you are giving it commands. (I'd need to confirm if its straight away after giving it command or a bit after)

Hypfer, 2022-01-23:
> What about the unknown propeties I found in the logs? Could that be related?

The unknown property change message is just the fan speed setting. Apparently these newer roborocks support both the old roborock command set as well as the newer miot-style commands.

See:
https://miot-spec.org/miot-spec-v2/instance?type=urn:miot-spec-v2:device:vacuum:0000A006:roborock-a15:1

No idea where that unknown_method thing is coming from but I doubt that it is relevant to this issue.

dicer, 2022-01-24:
I get this as well and I did the following:
Set a location on the map (close to the charging station) and click "Go to location". 
The state changes like this: `DOCKED` -> `MOVING - TARGET` -> `IDLE`
I get 2 errors.
The trace log contains this (I shortened some ids/ints):
```
2022-01-24T16:02:53.305Z
Debug
>>> cloud: {"method":"get_status","params":{},"id":1105}
2022-01-24T16:02:53.315Z
Debug
>>> cloud: {"method":"get_status","params":{},"id":1106}
2022-01-24T16:02:53.343Z
Debug
<<< cloud: {"id":1105,"result":[{"msg_ver":2,"msg_seq":103,"state":8,"battery":100,"clean_time":842,"clean_area":15310000,"error_code":0,"map_present":1,"in_cleaning":0,"in_returning":0,"in_fresh_state":1,"lab_status":0,"water_box_status":0,"fan_power":104,"dnd_enabled":0,"map_status":253,"is_locating":0,"lock_status":0,"water_box_mode":201,"water_box_carriage_status":0,"mop_forbidden_enable":0,"adbumper_status":[0,0,0],"water_shortage_status":0,"dock_type":0,"dust_collection_status":0,"auto_dust_collection":1,"mop_mode":300,"debug_mode":0}],"exe_time":3}
2022-01-24T16:02:53.355Z
Debug
<<< cloud: {"id":1106,"result":[{"msg_ver":2,"msg_seq":104,"state":8,"battery":100,"clean_time":842,"clean_area":15310000,"error_code":0,"map_present":1,"in_cleaning":0,"in_returning":0,"in_fresh_state":1,"lab_status":0,"water_box_status":0,"fan_power":104,"dnd_enabled":0,"map_status":253,"is_locating":0,"lock_status":0,"water_box_mode":201,"water_box_carriage_status":0,"mop_forbidden_enable":0,"adbumper_status":[0,0,0],"water_shortage_status":0,"dock_type":0,"dust_collection_status":0,"auto_dust_collection":1,"mop_mode":300,"debug_mode":0}],"exe_time":3}
2022-01-24T16:02:53.374Z
Debug
>>> cloud: {"method":"get_status","params":{},"id":1107}
2022-01-24T16:02:53.407Z
Debug
<<< cloud: {"id":1107,"result":[{"msg_ver":2,"msg_seq":105,"state":8,"battery":100,"clean_time":842,"clean_area":15310000,"error_code":0,"map_present":1,"in_cleaning":0,"in_returning":0,"in_fresh_state":1,"lab_status":0,"water_box_status":0,"fan_power":104,"dnd_enabled":0,"map_status":253,"is_locating":0,"lock_status":0,"water_box_mode":201,"water_box_carriage_status":0,"mop_forbidden_enable":0,"adbumper_status":[0,0,0],"water_shortage_status":0,"dock_type":0,"dust_collection_status":0,"auto_dust_collection":1,"mop_mode":300,"debug_mode":0}],"exe_time":3}
2022-01-24T16:02:55.608Z
Debug
>>> cloud: {"method":"get_status","params":{},"id":1108}
2022-01-24T16:02:55.620Z
Debug
<<< cloud: {"id":1108,"result":[{"msg_ver":2,"msg_seq":106,"state":8,"battery":100,"clean_time":842,"clean_area":15310000,"error_code":0,"map_present":1,"in_cleaning":0,"in_returning":0,"in_fresh_state":1,"lab_status":0,"water_box_status":0,"fan_power":104,"dnd_enabled":0,"map_status":253,"is_locating":0,"lock_status":0,"water_box_mode":201,"water_box_carriage_status":0,"mop_forbidden_enable":0,"adbumper_status":[0,0,0],"water_shortage_status":0,"dock_type":0,"dust_collection_status":0,"auto_dust_collection":1,"mop_mode":300,"debug_mode":0}],"exe_time":3}
2022-01-24T16:03:01.213Z
Debug
<<< cloud* {"stamp":1643040181}
2022-01-24T16:03:01.215Z
Debug
>>> cloud* {"stamp":1643040181}
2022-01-24T16:03:10.697Z
Debug
>>> cloud: {"method":"get_status","params":{},"id":1109}
2022-01-24T16:03:10.711Z
Debug
>>> cloud: {"method":"get_status","params":{},"id":1110}
2022-01-24T16:03:10.717Z
Debug
<<< cloud: {"id":1109,"result":[{"msg_ver":2,"msg_seq":107,"state":8,"battery":100,"clean_time":842,"clean_area":15310000,"error_code":0,"map_present":1,"in_cleaning":0,"in_returning":0,"in_fresh_state":1,"lab_status":0,"water_box_status":0,"fan_power":104,"dnd_enabled":0,"map_status":253,"is_locating":0,"lock_status":0,"water_box_mode":201,"water_box_carriage_status":0,"mop_forbidden_enable":0,"adbumper_status":[0,0,0],"water_shortage_status":0,"dock_type":0,"dust_collection_status":0,"auto_dust_collection":1,"mop_mode":300,"debug_mode":0}],"exe_time":3}
2022-01-24T16:03:10.732Z
Debug
<<< cloud: {"id":1110,"result":[{"msg_ver":2,"msg_seq":108,"state":8,"battery":100,"clean_time":842,"clean_area":15310000,"error_code":0,"map_present":1,"in_cleaning":0,"in_returning":0,"in_fresh_state":1,"lab_status":0,"water_box_status":0,"fan_power":104,"dnd_enabled":0,"map_status":253,"is_locating":0,"lock_status":0,"water_box_mode":201,"water_box_carriage_status":0,"mop_forbidden_enable":0,"adbumper_status":[0,0,0],"water_shortage_status":0,"dock_type":0,"dust_collection_status":0,"auto_dust_collection":1,"mop_mode":300,"debug_mode":0}],"exe_time":4}
2022-01-24T16:03:10.781Z
Debug
>>> cloud: {"method":"app_goto_target","params":[25270,25660],"id":1111}
2022-01-24T16:03:10.789Z
Debug
<<< cloud: {"id":1111,"result":["ok"],"exe_time":1}
2022-01-24T16:03:10.822Z
Debug
>>> cloud: {"method":"get_status","params":{},"id":1112}
2022-01-24T16:03:10.832Z
Debug
<<< cloud: {"id":1112,"result":[{"msg_ver":2,"msg_seq":109,"state":8,"battery":100,"clean_time":842,"clean_area":15310000,"error_code":0,"map_present":1,"in_cleaning":0,"in_returning":0,"in_fresh_state":1,"lab_status":0,"water_box_status":0,"fan_power":104,"dnd_enabled":0,"map_status":253,"is_locating":0,"lock_status":0,"water_box_mode":201,"water_box_carriage_status":0,"mop_forbidden_enable":0,"adbumper_status":[0,0,0],"water_shortage_status":0,"dock_type":0,"dust_collection_status":0,"auto_dust_collection":1,"mop_mode":300,"debug_mode":0}],"exe_time":2}
2022-01-24T16:03:11.693Z
Debug
[... cut ...]
>>> cloud: {"method":"get_map_v1","id":1129}
2022-01-24T16:13:38.474Z
Debug
<<< cloud: {"id":1129,"result":["retry"],"exe_time":1}
2022-01-24T16:13:38.551Z
Debug
Map upload started with: { query: { '...950': '' }, params: { filename: 'room_map_2' } }
2022-01-24T16:13:41.217Z
Debug
<<< cloud* {"stamp":1643040821}
```
I then sent it back to the station and got the following states (and 2 more `No error`)
`Returning - Resumable` -> `ERROR` -> `DOCKED - RESUMEABLE` -> `DOCKED`

Maybe this helps? Let me know if I can give you more info

adminy, 2022-01-25:
I heard mine say "Standby: Positioning failed" when I got the error.

Hypfer, 2022-02-04:
Previously, I've said the following:

> This is happening because the robot is in state 12 which is error

That however is not true. There's another status code (101) which maps to error as well.
Do we have a trace log from the error event to figure out which one the robot is actually reporting?

Hypfer, 2022-02-04:
Please set the loglevel to trace and then just search for the time around when the error is reported

adminy, 2022-02-04:
```
2022-02-04T15:40:07.349Z
Info
Set Logfile to /tmp/valetudo.log
2022-02-04T15:40:07.353Z
Trace
Trying to open device.conf at /mnt/default/device.conf
2022-02-04T15:40:07.357Z
Trace
Trying to open device.conf at /mnt/default/device.conf
2022-02-04T15:40:07.359Z
Trace
Trying to open device.conf at /mnt/default/device.conf
2022-02-04T15:40:07.361Z
Trace
Trying to open device.conf at /mnt/default/device.conf
2022-02-04T15:40:07.362Z
Trace
Trying to open device.conf at /mnt/default/device.conf
2022-02-04T15:40:07.364Z
Trace
Trying to open device.conf at /mnt/default/device.conf
2022-02-04T15:40:07.366Z
Trace
Trying to open device.conf at /mnt/default/device.conf
2022-02-04T15:40:07.368Z
Trace
Trying to open device.conf at /mnt/default/device.conf
2022-02-04T15:40:07.369Z
Info
Autodetected RoborockS7ValetudoRobot
2022-02-04T15:40:07.400Z
Trace
Trying to read token file at /mnt/data/miio/device.token
2022-02-04T15:40:07.413Z
Debug
>>> local: HandshakePacket()
2022-02-04T15:40:07.424Z
Trace
Trying to open device.conf at /mnt/default/device.conf
2022-02-04T15:40:07.426Z
Trace
Trying to open device.conf at /mnt/default/device.conf
2022-02-04T15:40:07.493Z
Info
Starting Valetudo 2022.02.0
2022-02-04T15:40:07.497Z
Info
Commit ID: 2d94cc3758d478b55b6245719f2c6d258c4b004f
2022-02-04T15:40:07.497Z
Info
Configuration file: /mnt/data/valetudo_config.json
2022-02-04T15:40:07.498Z
Info
Logfile: /tmp/valetudo.log
2022-02-04T15:40:07.499Z
Info
Robot: Roborock S7 (RoborockS7ValetudoRobot)
2022-02-04T15:40:07.499Z
Info
JS Runtime Version: v16.10.0
2022-02-04T15:40:07.500Z
Info
Arch: arm
2022-02-04T15:40:07.502Z
Info
Max Heap Size: 34 MiB
2022-02-04T15:40:07.502Z
Info
Node Flags: --expose-gc --max-heap-size=34 --optimize-for-size --lite-mode --no-regexp-tier-up --no-expose-wasm
[... 579 lines cut ...]
2022-02-04T15:47:08.733Z
Debug
<<< cloud: {"id":36,"result":["room_map_0"],"exe_time":1}
2022-02-04T15:47:08.779Z
Debug
Map upload started with: { query: { '68351207200': '' }, params: { filename: 'room_map_1' } }
2022-02-04T15:47:12.738Z
Debug
>>> cloud: {"method":"get_map_v1","id":37}
2022-02-04T15:47:12.748Z
Debug
<<< cloud: {"id":37,"result":["room_map_1"],"exe_time":1}
2022-02-04T15:47:12.794Z
Debug
Map upload started with: { query: { '68351287742': '' }, params: { filename: 'room_map_2' } }
2022-02-04T15:47:16.754Z
Debug
>>> cloud: {"method":"get_map_v1","id":38}
2022-02-04T15:47:16.763Z
Debug
<<< cloud: {"id":38,"result":["room_map_2"],"exe_time":1}
2022-02-04T15:47:16.821Z
Debug
Map upload started with: { query: { '68351351033': '' }, params: { filename: 'room_map_3' } }
```

Hypfer, 2022-02-04:
There are neither state 12 nor state 101 in that log 🤔

adminy, 2022-02-04:
I get this id 0 quite often ...

mehhhhhhhhhhhhhhhhhhhhhhhhhh, 2022-03-18:
Here's what I’m seeing under trace.  

[image]

KayGundhardt, 2022-03-19:
Hello everybody,

found out today that these errors do NOT occur when running the vacuum via MQTT-commands…

Does this help somehow?

mehhhhhhhhhhhhhhhhhhhhhhhhhh, 2022-04-15:
Given that the S7 is throwing code 0, which is apparently "No Error", why not just have the if statement not publish the error - no error message?

`const ERROR_CODES = {
    0: "No error",
    1: "LDS jammed",
    2: "Stuck front bumper",
    3: "Wheel lost floor contact. Robot is on the verge of falling",
    4: "Cliff sensor dirty or robot on the verge of falling",
    5: "Main brush jammed",
    6: "Side brush jammed",
    7: "Wheel jammed",
    8: "Robot stuck or trapped",
    9: "Dustbin missing",
    10: "Filter jammed",
    11: "Magnetic interference",
    12: "Low battery",
    13: "Charging issues",
    14: "Battery temperature out of operating range",
    15: "Wall sensor dirty",
    16: "Tilted robot",
    17: "Side brush error. Reboot required",
    18: "Fan error. Reboot required",
    19: "Charging station without power",
    21: "LDS bumper jammed",
    22: "Charging contacts dirty",
    23: "Charging station dirty",
    24: "Stuck inside restricted area",
    25: "Camera dirty",
    26: "Wall sensor dirty",
    29: "Animal excrements detected"

    //TODO: there are also 100+ codes. No idea when they might appear though
};

const GET_ERROR_CODE_DESCRIPTION = (errorCodeId) => {
    if (ERROR_CODES[errorCodeId] !== undefined) {
        return ERROR_CODES[errorCodeId];
    } else {
        return "UNKNOWN ERROR CODE " + errorCodeId;
    }
};`

Hypfer, 2022-04-18:
I am still waiting for someone to answer the question if it is status code 12 or 101 that is being reported by the robot

KayGundhardt, 2022-04-18:
Hello again,

I tried to get the reported status. The errors occurred today repeatedly between 19:36 and 19:38:

[image]

But as far as I can see, the traced log shows no entry between 17:50 and 20:28 :

```
2022-04-18T17:50:11.610Z
Debug
<<< cloud: {
  id: 6885,
  result: [
    {
      main_brush_work_time: 434794,
      side_brush_work_time: 641528,
      filter_work_time: 434794,
      filter_element_work_time: 0,
      sensor_dirty_time: 65990,
      dust_collection_work_times: 0
    }
  ],
  exe_time: 2
}
2022-04-18T17:50:11.625Z
Debug
>>> cloud: { method: 'get_status', params: {}, id: 6886 }
2022-04-18T17:50:11.637Z
Debug
<<< cloud: {
  id: 6886,
  result: [
    {
      msg_ver: 2,
      msg_seq: 3956,
      state: 8,
      battery: 100,
      clean_time: 17,
      clean_area: 0,
      error_code: 0,
      map_present: 1,
      in_cleaning: 0,
      in_returning: 0,
      in_fresh_state: 1,
      lab_status: 1,
      water_box_status: 1,
      fan_power: 104,
      dnd_enabled: 0,
      map_status: 3,
      is_locating: 0,
      lock_status: 0,
      water_box_mode: 203,
      water_box_carriage_status: 1,
      mop_forbidden_enable: 1,
      adbumper_status: [ 0, 0, 0 ],
      water_shortage_status: 0,
      dock_type: 0,
      dust_collection_status: 0,
      auto_dust_collection: 1,
      mop_mode: 300,
      debug_mode: 0
    }
  ],
  exe_time: 3
}
2022-04-18T20:28:46.061Z
Debug
<<< cloud* { stamp: [1650313725](tel:1650313725) }
2022-04-18T20:28:46.095Z
Debug
>>> cloud* { stamp: [1650313725](tel:1650313725) }
```

Maybe I did something wrong? Have chosen “trace” as current level from the Log and clicked on the symbol to refresh (open circle with arrow on the top right of the webfrontend Log view) - right?

Kay

Hypfer, 2022-04-19:
The loglevel setting is not a filter but a setting.
You need to first set the loglevel to trace and then wait for the issue to happen. Then it should also appear in the log.

KayGundhardt, 2022-04-19:
Okay! Hope I got it: it's the right one now? 
(4 "NoError"-Errors occured between 6:56 and 6:58)

```
2022-04-19T08:49:04.285Z
Debug
SSDP: Received M-SEARCH request for upnp:rootdevice from 10.0.1.2:49683
2022-04-19T08:49:04.467Z
Debug
>>> cloud: { method: 'get_status', params: {}, id: 3104 }
2022-04-19T08:49:04.477Z
Debug
<<< cloud: {
  id: 3104,
  result: [
    {
      msg_ver: 2,
      msg_seq: 1782,
      state: 8,
      battery: 100,
      clean_time: 17,
      clean_area: 0,
      error_code: 0,
      map_present: 1,
      in_cleaning: 0,
      in_returning: 0,
      in_fresh_state: 1,
      lab_status: 1,
      water_box_status: 1,
      fan_power: 104,
      dnd_enabled: 0,
      map_status: 3,
      is_locating: 0,
      lock_status: 0,
      water_box_mode: 203,
      water_box_carriage_status: 1,
      mop_forbidden_enable: 1,
      adbumper_status: [ 0, 0, 0 ],
      water_shortage_status: 0,
      dock_type: 0,
      dust_collection_status: 0,
      auto_dust_collection: 1,
      mop_mode: 300,
      debug_mode: 0
    }
  ],
  exe_time: 3
}
2022-04-19T08:49:04.589Z
Debug
>>> cloud: { method: 'get_consumable', params: [], id: 3105 }
2022-04-19T08:49:04.598Z
Debug
<<< cloud: {
  id: 3105,
  result: [
    {
      main_brush_work_time: 434811,
      side_brush_work_time: 641545,
      filter_work_time: 434811,
      filter_element_work_time: 0,
      sensor_dirty_time: 66007,
      dust_collection_work_times: 0
    }
  ],
  exe_time: 2
}
2022-04-19T08:49:04.614Z
Debug
>>> cloud: { method: 'get_status', params: {}, id: 3106 }
[... 1378 lines cut ...]
    did: '391203896',
    siid: 8,
    eiid: 1,
    arguments: [ { piid: 3, value: 0 } ]
  }
}
2022-04-19T08:51:41.221Z
Debug
>>> cloud: { id: 86368210, result: 'ok' }
2022-04-19T08:51:41.226Z
Debug
<<< cloud: {
  id: 145092404,
  method: 'properties_changed',
  params: [
    { did: '391203896', siid: 2, piid: 1, value: 8 },
    { did: '391203896', siid: 2, piid: 2, value: 0 },
    { did: '391203896', siid: 2, piid: 3, value: 104 },
    { did: '391203896', siid: 3, piid: 1, value: 100 }
  ]
}
2022-04-19T08:51:41.238Z
Debug
>>> cloud: { id: 145092404, result: 'ok' }
```

Hypfer, 2022-04-19:
Okay, I've found the issue

```
2022-04-19T08:51:14.185Z
Debug
<<< cloud: {
  id: 180973539,
  method: 'event_occured',
  params: {
    did: '391203896',
    siid: 8,
    eiid: 1,
    arguments: [ { piid: 3, value: 0 } ]
  }
}
```

The event handling code is wrong and doesn't check for the value of that piid.
As that was added by some PR and only applys to roborock robots that I don't have, I had completely forgotten that this codepath even exists.

Fixed with f3b4a548dffc3aa96c2ecdf37c063b73847b0d5e

closed 2022-04-19 by Hypfer