Hypfer/Valetudo #1285: Not able to update

macolinob, 2021-12-17:
Updating is impossible because there's not enough space to store the new binary at /usr/local/bin/valetudo. Required: 41943040 bytes. Available: 27691008 bytes.

I looked through the issues section and did not see anyone post about this.  The Roborock S5 is on Valetudo 2021.12.0.  Nothing stands on it the logs regarding the not enough space.

I did a reboot and tried again with no luck.

Can I try to flash manually and get past this point?

Thank you,

Bret

closed 2021-12-17 by Hypfer

macolinob, 2021-12-17:
Did I miss something?

macolinob, 2021-12-17:
I did read in the upgrade notes that the best option was to do the manual way.  I do apologize for that, I just was letting everyone know that the "Updater" did work once but not the second time to  2021.12.1

Thank you for your time,

Bret

Hypfer, 2021-12-17:
https://github.com/Hypfer/Valetudo/issues/1282
https://github.com/Hypfer/Valetudo/issues/1234
https://github.com/Hypfer/Valetudo/discussions/1272

> I looked through the issues section and did not see anyone post about this

No, you obviously didn't.

You also opened an issue when everything clearly states that you shall not open issues but use the discussion feature.

macolinob, 2021-12-17:
I apologize I also did not look through closed issues.  Please excuse my ignorance.