Hypfer/Valetudo #1295: Map resets on Dreame L10 Pro

meks007, 2021-12-21:
Hi,

I have updated from 2021.11.0 to 2021.12.1 and I'm now facing issues when attempting to modify segments created by e.g. mapping runs or previous cleanups.

1. Reset persistent data
2. Do a mapping run
3. Robot does not create segments correctly -> Edit Map
4. Select a single big segment 
5. Insert and position cutting line
6. Click Split

Map does not update, nor does it appear to save the changes after clicking Save etc. The map editor does not reflect any manual changes, however when attempting to do the same split again, different error messages appear depending on the attempted split:
* Segment too small
* Segment not within walls

After restarting Valetudo using ssh/reboot or ssh/killall valetudo & valetudo SOMETIMES the map loads and displays previous modifications OR more often  the map resets and Valetudo shows "No Map Data".

I'm only facing this issue on the L10. The S5 works completely normal.

closed 2021-12-21 by Hypfer

meks007, 2021-12-21:
Is there a reason why this is invalid?

Hypfer, 2021-12-21:
You're blatantly ignoring the rules of this issue tracker.

[image]

meks007, 2021-12-21:
Aha! Well, no, not blatantly ignoring. I was not presented an issue template when filing the issue the first time. This is a bug on Github's side, when you start the issue without being logged in. Github asks you to log in but doesn't present the template afterwards.

I'll re-do it.