Hypfer/Valetudo #1296: Map resets after using Map Editor on Dreame L10 Pro

meks007, 2021-12-21:
## Describe the bug

I have updated from 2021.11.0 to 2021.12.1 and I'm now facing issues when attempting to modify segments created by e.g. mapping runs or previous cleanups.

After editing the map does not update, nor does it appear to save the changes after clicking Save etc. The map editor does not reflect any manual changes, however when attempting to do the same split again, different error messages appear depending on the attempted split:

* Segment too small
* Segment not within walls

After restarting Valetudo using ssh/reboot or ssh/killall valetudo & valetudo SOMETIMES the map loads and displays previous modifications OR more often the map resets and Valetudo shows "No Map Data".

I'm only facing this issue on the L10. The S5 works completely normal.
## To Reproduce

1. Reset persistent data
2. Do a mapping run
3. Robot does not create segments correctly -> Edit Map
4. Select a single big segment
5. Insert and position cutting line
6. Click Split
### Screenshots

none

### Vacuum Model

Dreame L10 Pro

### Valetudo Version

2021.12.1

## Expected behavior

Manual changes to the map are reflected in the map editor instead of leading to a map reset.

## Additional context

none

Hypfer, 2021-12-21:
it's not a bug either..

closed 2021-12-21 by Hypfer

meks007, 2021-12-21:
Right then. I have brought this to your attention. I'll move on now :)