Hypfer/Valetudo #1306: Dustbin attachment status not working on Dreame F9

Schorsch99, 2021-12-23:
## Describe the bug
Dustbin attachment status does not change when dustbin is removed. Status is permanently "attached"

## To Reproduce

Steps to reproduce the behavior:
1. Go to Valetudo WebUI
2. View "Attachments" in WebUI and/or MQTT Status
3. Place/Remove dustbin
4. -> No dustbin Status Change (always "attached")
5. Place/remove watertank/mop
6. -> Status change of watertank/mop reported correctly

### Screenshots

### Vacuum Model

Dreame F9

### Valetudo Version

2021.12.1

## Expected behavior

Dustbin status change reported correctly in WebUI and MQTT according to attachment status

## Additional context

There is also no voice message when removing or placing the dustbin.

Hypfer, 2021-12-23:
The dustbin status unfortunately isn't reported by the robot.

closed 2021-12-23 by Hypfer

Schorsch99, 2021-12-23:
The factory firmware had voice messages when placing/removing dustbin, so it must report the state somewhere, I assume.

Hypfer, 2021-12-23:
If you find it let me know

Schorsch99, 2021-12-23:
Any hint where to start searching?

Hypfer, 2021-12-23:
I was being sarcastic.