Hypfer/Valetudo #1307: Manual control does not work anymore

EmilioDelCastillo, 2021-12-23:
## Describe the bug

After updating my robot from 0.6.1 to 2021.12.1, manual control only works for clockwise rotation, sometimes counter-clockwise but it's really not reliable. Forwards and backwards don't do anything.
When inspecting the requests, the robot returns 200 but does not execute the command.

## To Reproduce
Steps to reproduce the behavior:
1. Go to Manual Control
2. Enable manual control
3. Try to control the robot.

### Vacuum Model

Roborock V1
### Valetudo Version

2021.12.1
## Expected behavior

The robot should be able to be controlled in all directions.
## Additional context

It also surprised me that it now only seems to work by increments instead of a continuous motion as it was with 0.6.1, where we could control the speed of the movement.

bademux, 2022-01-15:
the symptoms are quite funny - if you press bumper (vacuum feels smth in front of him) it moves forward

satrik, 2022-02-12:
I have exactly the same behavior with my Roborock V1 and Valetudo 2022.02.0....

Hypfer, 2022-04-10:
abccee47

closed 2022-04-10 by Hypfer