Hypfer/Valetudo #1321: WebUI not accessible anymore after update

Naabmaster69, 2022-01-01:
## Web UI error: Valetduo blocks access from external public IPs

## To Reproduce: Try to access WEB UI with smart phone after Update

### Dreame Z10

### Valetudo Version 2021.1.0

## Pls help. Reboot didn‘t help, how can I fix this issue?

Naabmaster69, 2022-01-01:
I tried to access the robot over my private network, no VPN or sth in between

Hypfer, 2022-01-01:
Valetudo uses the `is-in-subnet` library and allows access from either localhost or all private IP ranges.

See here:
https://github.com/natesilva/is-in-subnet/blob/1a9da7b8713fe3e7db0eaf6398dca9f2fa83f651/src/ipRange.ts#L3-L25

If you're not able to access your robot your network is likely misconfigured.

Naabmaster69, 2022-01-01:
True, I‘m not using the standard private ip address. Is there a way hiw to access the WebUI again wirhout changing my network ip address?

Hypfer, 2022-01-01:
Please reconfigure your network to use a non-publicly-routable address range then.
I'm fairly certain that the way it is now already causes a lot of breakage in other applications.

closed 2022-01-01 by Hypfer