Hypfer/Valetudo #1324: Unable to add virtual restrictions on Dreame 1C

sitway, 2022-01-03:
## Describe the bug

Adding virtual restrictions on the map (such as no go area or virtual wall) didn't seem to work on Dreame 1C.
Whenever I tried to add a virtual wall or no go area it will always disappear once I click the save button. I tried clicking the refresh button or reloading the page but it didn't work either, there weren't any markings on the map.  
I've tried multiple times with different browser, three different versions of valetudo (2022.01.0, 2021.12.1, 2021.11.1),  two different firmware versions (1096 and 1110) and still getting the same results.
I've checked the log and didn't find any warnings or errors related to what I was trying to do.

P.S: The robot was in docked state and had already completed cleaning and successfully returned to the dock by itself.

## To Reproduce

1. Go to 'Edit Map'
2. Click on 'Edit Virtual Restrictions'
3. Click on  'Add virtual wall' or 'Add no go area'
4. Place it anywhere on the map
5. Hit save
6. Boom! now it's gone
7. Try to hit refresh button and it will be still the same

### Vacuum Model

Dreame 1C

### Valetudo Version

2022.01.0, 2021.12.1, 2021.11.1

## Expected behavior

It should work according to this [page](https://valetudo.cloud/pages/general/supported-robots.html#dreame_1c)

Hypfer, 2022-01-03:
I know that it works because I've tested it with my own 1C.

Persistent maps need to be enabled

closed 2022-01-03 by Hypfer

sitway, 2022-01-03:
Persistent maps is already enabled from the beginning, what firmware version are you using ?

ibanmda, 2022-01-11:
I have same issue. Now I can't reset map of my 1C. When I try do it, the UI show continuos loop in persistent data, but does not finish with succesfuly. I can't delete virtual barrier :(

**Dreame1CValetudoRobot
Valetudo 2022.01.0**