Hypfer/Valetudo #1325: Not only external traffic is blocked

quixony, 2022-01-03:
I updated via the Updater feature to the new 2022.01.1. But there was no "Successful" dialog. Instead it shows an error message that no information could be fetched. Then it is no longer possible to access the webUI at all.
My internal network running at 192.167.177.0/24 (due to vpn connected fritzbox routers).

## To Reproduce
Get a network environment not having the default 192.168.178.0/24 ip range. Configure the roborock in 2021.12.1
Then update to 2022.1.1.

### Vacuum Model
Roborock S5

### Valetudo Version
2021.12.1 -> 2022.1.1

## Expected behavior
Every thing works as before. The robot would set its own IP subnet as "private" and every thing works.

rimrul, 2022-01-03:
> My internal network running at 192.167.177.0/24 

That's not a private IPv4 address range, so this behaviour is expected.

> To overcome the issue it is possible to log in via SSH and adjust the config file /mnt/data/valetudo_config.json and set the parameter webserver:blockExternalAccess: false. After a reboot the robo works as expected again 
> […] But this is not the general advice for everyone.

No, the general advice is to not use public address ranges for your LAN.

Hypfer, 2022-01-03:
Quoting from [RFC1918](https://datatracker.ietf.org/doc/html/rfc1918#section-3)

```
3. Private Address Space

   The Internet Assigned Numbers Authority (IANA) has reserved the
   following three blocks of the IP address space for private internets:

     10.0.0.0        -   10.255.255.255  (10/8 prefix)
     172.16.0.0      -   172.31.255.255  (172.16/12 prefix)
     192.168.0.0     -   192.168.255.255 (192.168/16 prefix)
```

`192.167.0.0/16` seems to belong to `AS137 Consortium GARR` btw, which - according to google - is `the Italian national computer network for universities and research.`

closed 2022-01-03 by Hypfer