Hypfer/Valetudo #1329: Valetudo remembers old MQTT identifier

alve89, 2022-01-04:
**First of all: Thank you - you're doing a really great job! I flashed my S6 today with [Dustbuilder](https://dustbuilder.dontvacuum.me) and I'm fascinated how properly the Valetudo software itself is working (including the map, rooms and so on). Both thumbs up!**

But now to my problem.

## Describe the bug
After setting up the MQTT settings the generic identifier is still used.

That's the generic identifier:
[image]

That's the identifier is used:
[image]

That's what MQTT Explorer recognizes:
[image]

That's what my MQTT broker (Mosquitto as Home Assistant add-on in HassOS) logs:
[image]

In my opinion `valetudo_ThatLightheartedCurlew` shouldn't be used anymore because I replaced it by `myVeryPrivateRoborock`...?

These are the steps I tried to solve the problem (multiple times, in different order):
- Restart Mosquitto
- Restart Home Assistant
- Restart Roborock (via SSH)
- Changing MQTT login credentials
- Changing MQTT identifier

Valetudo or Mosquitto still use the identifier `valetudo_ThatLightheartedCurlew` although it was changed to `myVeryPrivateRoborock`.

## To Reproduce

### Screenshots

This is my Valetudo MQTT configuration:
[image]

### Vacuum Model

Roborock S6

### Valetudo Version

[image]

## Expected behavior

I expect to see the same identifier (`myVeryPrivateRoborock`) everywhere and especially I don't see `valetudo_ThatLightheartedCurlew` somewhere after changing it.

## Additional context

I just rebooted the Roborock once more (after completely rebooting the whole RPi):
[image]

This shows that Valetudo is "logging in" to MQTT as `valetudo_ThatLightheartedCurlew`.

Could this also be the reason why the [Valetudo Map Card](https://github.com/TheLastProject/lovelace-valetudo-map-card) doesn't work?
[image]
[image]

Hypfer, 2022-01-04:
Are you sure that you're not just seeing retained messages?

Hypfer, 2022-01-04:
TL;DR:

The `MQTT ClientId` is unaffected by the identifier setting.

https://github.com/Hypfer/Valetudo/blob/5bab773230a4d51761b5ea3a9d1c274e4a94bbab/backend/lib/mqtt/MqttController.js#L122-L123

This is indeed true. I don't think it should be though, given that it only appears in the mqtt broker logs.
It might be a misunderstanding of the meaning of identifier.  The machine readable identifier is currently used to e.g. name the topics, name the home assistant device etc.

Problem is that the `MQTT ClientId` has to be unique as the mqtt broker will terminate the older connection if another device with the same client id appears because it assumes that it's the same device and the connection just broke

There will be at least three people misconfiguring this so that multiple robots fight over the same identifier. Then, there will be at least five issues complaining about Valetudos MQTT client being broken.

> Could this also be the reason why the Valetudo Map Card doesn't work?

No

alve89, 2022-01-05:
Alright, thank you for this detailed explanation. 
So this is an expected behavior as

- `valetudo_ThatLightheartedCurlew` is the `clientId` (which is auto-generated) and
- `myVeryPrivateRoborock` is only used for topics and so on.

Do you also know which of both have to be used for the Lovelace Map Card?
If yes, your answer will be very appreciated, if no, please just close this issue. 

Thanks again for your effort and help!

alve89, 2022-01-05:
I found the problem with the Lovelace Map Card. 
The configuration is
```
type: custom:valetudo-map-card
vacuum: valetudo_myveryprivateroborock
```

So it's the user-given robot name in **lowercase**.

closed 2022-01-05 by alve89