Hypfer/Valetudo #1333: Virtual restriction editing is not fully disabled when robot is not docked

supersmile2009, 2022-01-06:
## Describe the bug
Existing virtual restriction controls are not disabled when robot isn't docked. It gives user a false impression that existing restrictions can be edited while they can't be.

## To Reproduce
1. Make sure you have at least one virtual restriction.
2. Send robot to do something (cleaning, Go to location, etc.). Basically make sure it's not docked.
2. Go to Menu -> Map Management -> Virtual Restriction Management
3. Click on any restriction. You will see the "resize" and "remove" controls (which are active and take effect visually when used) and will be able to move the restriction around. You can't save the changes, because there's no "Save" button. But you get a false impression that you can edit this stuff if you don't notice the message in the corner saying that robot has to be docked.

### Screenshots
[image]

### Vacuum Model
Roborock S7, but it doesn't look like a model-specific bug.

### Valetudo Version
2022.01.0

## Expected behavior
All controls as well as ability to move virtual restrictions should be disabled if robot is not docked.

Hypfer, 2022-01-07:
That's not a bug. It's a feature request

Hypfer, 2022-01-07:
b68e44428b3c2669cae147c8c4f21bbfc9bc016d

closed 2022-01-07 by Hypfer

supersmile2009, 2022-01-08:
> That's not a bug. It's a feature request
It looked like an unintentional omission to me since the rest of the things were disabled.

Anyway thanks for quick response and for fixing/implementing it.